darrylb123/usbrelay #10: Interested to contribute to crelay?

ondrej1024, 2015-03-05:
Hi,

I just got notified of your project and was wondering if you would be interested to contribute to the crelay project:
http://ondrej1024.github.io/crelay

This software aims at providing a unified way of controlling different relay types (mostly USB). I wrote the framework and the drivers for 2 USB relay cards. It would be great if you wanted to add the driver for the HIDAPI type of relay card that your usbrelay software supports. 

Basically what you would need to write is the low level functions for this relay card to perform the following:
- detection of the card and communication port
- set relay states
- read relay states

Please let me know what you think about this.
Thanks, Ondrej

darrylb123, 2015-03-05:
Ondrej,
I don't have a problem with contributing code to your project.
I shall have a look at it.

Darryl

ondrej1024, 2015-03-06:
Hi Darryl, thanks for your interest. 
I had a closer look at your code and I think I understand now how the protocol works. How did you manage to discover the protocol, is there some datasheet for these HIDAPI relays around?

Could you please confirm the following description?

Relay Status Request
-----------------------------

Send the following buffer using
hid_get_feature_report():

Byte  1 2 3 4 5 6 7 8 9
Data 01 - - - - - - - -

The following buffer is received
as response:

Byte  1 2 3 4 5 6 7 8 9
Data  C C C C C 0 ? S ?

C: 5 characters which form the relay cards Identity string
S: status byte where each bit indicates the state of the corresponding relay

Relay state setting
-----------------------------

Send the following buffer using
hid_write()

Byte  1 2 3 4 5 6 7 8 9
Data 00 S R 0 0 0 0 0 0

S: Relay state (0xff=on, oxfd=off)
R: Relay number (integer)

darrylb123, 2015-03-06:
I sniffed the protocol with wireshark monitoring a Windows VM with the USB
device plugged through in kvm/virtmgr. The supplied windows software was
used.

I have to try to remember but here goes:
The number of relays are in  Product when you query the USB device, the
last character is the number of relays USBRelay2

You are correct with the status report

The state setting buffer is 0 S R 0 0 0 0 0 0

Do you have any of these devices, happy to test if not.

Darryl

ondrej1024, 2015-03-06:
Actually I don't have any of these relay cards so I would be glad if you could help testing. I will try to come up with a test version during the weekend.
Ondrej

ondrej1024, 2015-03-07:
Hi Darryl,

I have now a new version of crelay ready with support for HID API compatible relay cards. Multiple cards are not yet supported, the first device found will be used. It would be great if you could test this. You can get the software by cloning the hidapi_experimental branch from my GitHub repository:
git clone https://github.com/ondrej1024/crelay --branch hidapi_experimental

Before building, you might need to install the dependencies:
apt-get install libftdi1 libftdi-dev libhidapi-libusb0 libhidapi-dev

Then just go into the src directory and build:
cd crelay/src
make

Now you can run crelay. You might want to try "crelay -i" first to see if the relay card is detected (might need to run this as root). The syntax for switching the relays is as follows.
crelay [<relay number>] [ON|OFF]
Thanks, Ondrej

darrylb123, 2015-03-09:
> crelay [] [ON|OFF]

Yep, it works:
[dbond@gold src]$ sudo ./crelay 1 ON
DBG: card USBRelay2 found
DBG: portname 0003:0004:00
DBG: Write relay data FF 01
[dbond@gold src]$ sudo ./crelay 2 ON
DBG: card USBRelay2 found
DBG: portname 0003:0004:00
DBG: Write relay data FF 02
[dbond@gold src]$ sudo ./crelay 2 OFF
DBG: card USBRelay2 found
DBG: portname 0003:0004:00
DBG: Write relay data FD 02
[dbond@gold src]$ sudo ./crelay 1 OFF
DBG: card USBRelay2 found
DBG: portname 0003:0004:00
DBG: Write relay data FD 01

BTW, I use Fedora and libftdi on Fedora requires the following changes
diff --git a/src/Makefile b/src/Makefile
index 2beb759..92c496e 100644
--- a/src/Makefile
+++ b/src/Makefile
@@ -17,7 +17,7 @@ CC    = gcc
 INCLUDE        = -I.
 DEFS   = -D_GNU_SOURCE
 CFLAGS = $(DEBUG) $(DEFS) -Wformat=2 -Wall -Winline $(INCLUDE) -pipe -fPIC
-LIBS   = -lftdi -lhidapi-libusb
+LIBS   = -lftdi1 -lhidapi-libusb

 # Main source files (don't change)
 #SRC   = crelay.c relay_drv.c relay_drv_conrad.c relay_drv_gpio.c
diff --git a/src/relay_drv_sainsmart.c b/src/relay_drv_sainsmart.c
index 2640fcb..f33cde0 100644
--- a/src/relay_drv_sainsmart.c
+++ b/src/relay_drv_sainsmart.c
@@ -53,7 +53,7 @@
 #include <string.h>
 #include <unistd.h>
 #include <errno.h>
-#include <ftdi.h>
+#include <libftdi1/ftdi.h>

 #include "data_types.h"
 #include "relay_drv.h"

Just a comment, I went through a few iterations when wanted to support
multiple relay cards. I ended up using the format UNIQUE_RELAYID=STATE for
reading and writing.
I can eval the return STDOUT to capture all the relay states as shell
variables.
Makes it simple to do logic in the shell scripts then write out the new
values. There is a couple of examples in my github doco.

Darryl

ondrej1024, 2015-03-09:
Hi Darryl, thank you very much for testing. I'm glad (and somewhat surprised) it's already working. Could you also test the reading of the relay state please?
Just issue "sudo ./crelay <relay number>" without "on" or "off" and it should print the current state.

Too bad that different distributions have different names for the libs and put the header files in different locations. I don't really know how to solve this. I might probably use some flags to exclude certain relay drivers and their dependencies if you don't need them.

Ondrej

ondrej1024, 2015-03-09:
And what does "sudo ./crelay -i" print?

darrylb123, 2015-03-09:
> Just issue "sudo ./crelay " without "on" or "off" and it should print the

[dbond@gold src]$ sudo ./crelay 1
DBG: card USBRelay2 found
DBG: portname 0003:0005:00
DBG: Relay ID: 0U70M
DBG: Read relay bits 00
Relay 1 is off
[dbond@gold src]$ sudo ./crelay 1 ON
DBG: card USBRelay2 found
DBG: portname 0003:0005:00
DBG: Write relay data FF 01
[dbond@gold src]$ sudo ./crelay 1
DBG: card USBRelay2 found
DBG: portname 0003:0005:00
DBG: Relay ID: 0U70M
DBG: Read relay bits 01
Relay 1 is on

darrylb123, 2015-03-09:
[dbond@gold src]$ sudo ./crelay -i
DBG: card USBRelay2 found
DBG: portname 0003:0005:00
Detected relay card type is HID API compatible relay card (on 0003:0005:00)

ondrej1024, 2015-03-09:
Wow, that was quick. Thanks a lot.

darrylb123, 2015-03-09:
I'm in AU, this is a good time if you want interactive.

closed 2018-01-27 by darrylb123