darrylb123/usbrelay #11: Question: what about compiling/using this under Windows?

JamesWHurst, 2015-05-12:
I apologize - not quite sure how else to contact you other than submitting this question as an Issue.

Can this library be used to operate these relay switches under Windows?  I am specifically running Windows 7 embedded, for a target machine, and Windows 7 and 8.1 on my dev boxes.

And too -- thank you for providing this for others to share and utilize. I appreciate your work.

James Hurst

darrylb123, 2015-05-12:
As far as I can tell, the hidapi library is available for Windows.
http://www.signal11.us/oss/hidapi/

I expect it should be able to be compiled for windows,  I haven't used
linux specific libraries.
I have no capability of trying it though.

I would be interested in knowing if it does.
Patches welcome.

Darryl

closed 2018-01-27 by darrylb123

pavolgaj, 2022-03-21:
Did someone successfully compile it on Windows? I tried MinGW but I couldn't link hidapi lib.

darrylb123, 2022-03-21:
Not that I am aware of.
Have you compiled the hidapi library? Looks like Windows is one of the
targets.
https://github.com/libusb/hidapi

pavolgaj, 2022-03-22:
I directly downloaded hidapi release for Windows with compiled dll. And copy everything to the same folder with usbrelay files.

darrylb123, 2022-03-22:
I know nothing about compiling for windows.
I assume you could adjust the makefile for the dll instead of the so.

You would need to set the library location instead of the default. I think
gcc uses the -L to set the library Path. Toy would also need to set the
include path to where hidapi includes are.

You probably should find someone that has done something similar.

Darryl

jacobq, 2022-06-05:
See https://github.com/pavel-a/usb-relay-hid/releases/tag/usb-relay-lib_v2.1

pavolgaj, 2022-06-05:
The big advantage of the usbrelay code is the possibility to change relay ID. This feature isn't implemented in USB Relay Hid...

darrylb123, 2022-06-05:
Can the supplied windows software change the USB serial.
I originally used that on a VM then sniffed the USB bus to determine the
protocol.
Haven't looked at it for many years though. The source for firmware on the
card is on GitHub. That is how I discovered how to change the serial.

What about running a Linux VM on windows to change the id.

jacobq, 2022-06-05:
> The big advantage of the usbrelay code is the possibility to change relay ID. This feature isn't implemented in USB Relay Hid...

Thanks for noting that. The reason I linked the other (older and perhaps inferior) project is because it has a simply GUI test program that runs in Windows (precompiled binary available for download).