darrylb123/usbrelay #14: Compile in linux undefined reference error

gokcer, 2017-03-13:
Thank you for creating this repository and sharing your valuable experience.

When I tried to compile repo using make, I got these errors:

`cc -O2 -Wall  -lhidapi-hidraw  usbrelay.c   -o usbrelay
/tmp/ccU9yOsi.o: In function `operate_relay':
usbrelay.c:(.text+0x4d): undefined reference to `hid_write'
usbrelay.c:(.text+0x8f): undefined reference to `hid_error'
/tmp/ccU9yOsi.o: In function `main':
usbrelay.c:(.text.startup+0xc5): undefined reference to `hid_enumerate'
usbrelay.c:(.text.startup+0x1cb): undefined reference to `hid_open_path'
usbrelay.c:(.text.startup+0x1f0): undefined reference to `hid_get_feature_report'
usbrelay.c:(.text.startup+0x34f): undefined reference to `hid_close'
usbrelay.c:(.text.startup+0x53f): undefined reference to `hid_free_enumeration'
usbrelay.c:(.text.startup+0x544): undefined reference to `hid_exit'
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'usbrelay' failed
make: *** [usbrelay] Error 1`

I already installed hidapi-hidraw library using apk manager.

I am a bit unexperienced on linux compilation. How can I continue.

Thanks.

darrylb123, 2017-03-13:
Apk? I'm not familiar with compiling for Android.
Compiling for Linux like Debian or Fedora requires the development
packages.

gokcer, 2017-03-13:
Oh so sorry :D I mean apt manager. Ubuntu package manager. I tried to mention I installed precompiled versions of the dependent libraries.

darrylb123, 2017-03-13:
Just check, there are precompiled packages for usbrelay in Debian. Try
apt-get install usbrelay

You will need to install the hidapi-dev packages if you want to compile it
yourself.

gokcer, 2017-03-13:
Thank you for the precompiled version. I didn't know that. By the way I had installed dev libraries for hidapi

`libhidapi-dev is already the newest version (0.8.0~rc1+git20140818.d17db57+dfsg-1).`

Still taking the same error. To clarify more

- I cloned the repository
- Installed libhidapi-dev by `# apt-get install libhidapi-dev`
- Changed directory to usbrelay
- Executed `# make`

Am I missing anything?

K1773R, 2017-03-13:
the issue with compiling is that you have to list the libs at the end, ie:
```
cc -O2 -Wall usbrelay.c -o usbrelay -lhidapi-hidraw
```

gokcer, 2017-03-13:
Thank you very much @K1773R . It fixed the problem but why `make` didn't worked out of the box? I inspected MakeFile to create a pull request but it seems okay.

```
CFLAGS = -O2 -Wall
HIDAPI = hidraw
LDFLAGS = -lhidapi-$(HIDAPI)

all: usbrelay

clean:
        rm -f usbrelay

install: usbrelay
        install -d $(DESTDIR)/usr/bin
        install -m 0755 usbrelay $(DESTDIR)/usr/bin

.PHONY: all clean install
```

So how can we fix this issue for the others not to have problem?

gokcer, 2017-03-13:
@K1773R @darrylb123 Thank you very much for all your precious time and work. It is compiling after @K1773R 's suggestion. @darrylb123  I think it will be good to accept the waiting pull request for the community.

closed 2017-03-14 by gokcer