darrylb123/usbrelay #15: I've got a USBRelay 4 but it works only on the last 2 Relays

pvangorp, 2017-04-05:
**Everything seems to work, but when I switch HW341_1/HW341_2 the relay won't click and the led doesn't work. The software recognizes the device correctly.**
# sudo ./usbrelay
Device Found
  type: 16c0 05df
  path: 0001:0005:00
  serial_number: (null)
  Manufacturer: www.dcttech.com
  Product:      USBRelay4
  Release:      100
  Interface:    0
  Number of Relays = 4
HW341_1=0
HW341_2=0
HW341_3=0
HW341_4=0

**So when I execute "# sudo ./usbrelay HW341_1=1 HW341_2=1 HW341_3=1 HW341_4=1" I get:**
Orig: HW341, Serial: HW341, Relay: 1 State: ff
Orig: HW341, Serial: HW341, Relay: 2 State: ff
Orig: HW341, Serial: HW341, Relay: 3 State: ff
Orig: HW341, Serial: HW341, Relay: 4 State: ff
Device Found
  type: 16c0 05df
  path: 0001:0002:00
  serial_number: (null)
  Manufacturer: www.dcttech.com
  Product:      USBRelay4
  Release:      100
  Interface:    0
  Number of Relays = 4
Serial: HW341, Relay: 1 State: ff
1 HID Serial: HW341 Serial: HW341, Relay: 1 State: ff
Serial: HW341, Relay: 2 State: ff
2 HID Serial: HW341 Serial: HW341, Relay: 2 State: ff
Serial: HW341, Relay: 3 State: ff
3 HID Serial: HW341 Serial: HW341, Relay: 3 State: ff
Serial: HW341, Relay: 4 State: ff
4 HID Serial: HW341 Serial: HW341, Relay: 4 State: ff

Serial: HW341, Relay: 1 State: ff --- Found
Serial: HW341, Relay: 2 State: ff --- Found
Serial: HW341, Relay: 3 State: ff --- Found
Serial: HW341, Relay: 4 State: ff --- Found

**But only relay 3 and 4 are switched on an have a red LED. Relay 1 and 2 don't click and LED stay off.
Is this a bug in the code or is my relay defect?
Can you help me out please!**

pvangorp, 2017-04-05:
**In addition a full dump of lsusb -v -d 16c0:05df**

Bus 001 Device 002: ID 16c0:05df Van Ooijen Technische Informatica HID device except mice, keyboards, and joysticks
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0
  bDeviceProtocol         0
  bMaxPacketSize0         8
  idVendor           0x16c0 Van Ooijen Technische Informatica
  idProduct          0x05df HID device except mice, keyboards, and joysticks
  bcdDevice            1.00
  iManufacturer           1
  iProduct                2
  iSerial                 0
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           34
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0x80
      (Bus Powered)
    MaxPower               20mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      0 None
      iInterface              0
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.01
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      22
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval             100

darrylb123, 2017-04-05:
I'd say there is some kind of physical fault. The fact that 2 work and 2
don't indicates that. That said, I've never tested a 4 relay module. Can
you test with the Windows software?
Can you warranty it?

Darryl

pvangorp, 2017-04-05:
Thanks for you quick response. It's a new device so I have warranty. How do I test in Windows? I don't know how to test it in Visual Studio 2015. Can you help me? I can program a bit of C#. But I don't know were to get a DLL to talk to the device. Or how to...

darrylb123, 2017-04-05:
No, there is a demo program available from whatever site you bought if
from. It is already compiled and ready to go.

pvangorp, 2017-04-05:
I downloaded hidapi-0.7.0.zip  the Visual Studio 2010 code from https://github.com/signal11/hidapi/downloads. But the VID & PID are different. Is changed them, but can get feature report from the device. https://www.sossolutions.nl/4xusbrelais
They link to FTDI version from another brand. So that's not gone work. But your code partly. Isn't because your code is for a 2 relay?

darrylb123, 2017-04-05:
This eBay item shows the UI of a windows application that is supplied with
the device. http://www.ebay.com.au/itm/281203054240
You should be able to get a copy from where you bought it. No need for
visual studio.

pvangorp, 2017-04-05:
Thank you for your help. I tested it with their code and same issue. Only two work not 4. I'll call the store tomorrow.

twinee, 2017-06-05:
hey pvangorp, i've got the same board from sos solutions, not having much luck with it. did you get it to work? and if so, could you please pass me the code?

pvangorp, 2017-06-05:
No problem. I use the device in combination with a Raspberry Pi3 and Pi Zero W. The steps I took.
**Install the pre requirements**
# sudo apt-get install python-ftdi python-pip
# sudo pip install pylibftdi
# python -m pylibftdi.examples.list_devices
**_Bus 001 Device 003: ID XXXX:XXXX Van Ooijen Technische Informatica HID device except mice, keyboards, and joysticks_**
# sudo apt-get update
# sudo apt-get upgrade
# ls
# mkdir usbrelay
# cd usbrelay/
# git init
# git clone https://github.com/darrylb123/usbrelay.git
# cd ..
# mkdir hidapi
# cd hidapi/
# git init
# git clone https://github.com/signal11/hidapi.git
# sudo apt-get install libudev-dev libusb-1.0-0-dev libfox-1.6-dev autotools-dev autoconf automake libtool
# cd hidapi/
# ./bootstrap
# ./configure
# make
# sudo make install
# cd ~/usbrelay/usbrelay/
# gcc -o usbrelay usbrelay.c -lhidapi-hidraw
# gcc -o usbrelay usbrelay.c -lhidapi-libusb
# sudo ./usbrelay
**_Device Found
  type: XXXX XXXX
  path: XXXX:XXXX:XX
  serial_number: (null)
  Manufacturer: www.dcttech.com
  Product:      USBRelay4
  Release:      100
  Interface:    0
  Number of Relays = 4
HW341_1=0
HW341_2=0
HW341_3=0
HW341_4=0_**

**To get the relay state**
# sudo ./usbrelay 2>/dev/null

**To use the state in a script**
# eval $(sudo ./usbrelay 2>/dev/null)

**For relay 1**
# echo $ HW341_1
Returns state e.g. 0/1

**For relay 2**
# echo $ HW341_2

**For relay 3**
# echo $ HW341_3

**For relay 4**
# echo $ HW341_4

**Switch Relay 1 On**
# sudo ./usbrelay HW341_1=1

**Switch Relay 1 Off**
# sudo ./usbrelay HW341_1=0

**To set the relay state of 1 or more modules at once**
# sudo ./usbrelay HW341_1=0
# sudo ./usbrelay HW341_2=1 HW341_1=0
# sudo ./usbrelay HW341_2=0 HW341_1=1 0U70M_1=0 0U70M_2=1

**All relays on**
# sudo ./usbrelay HW341_1=1 HW341_2=1 HW341_3=1 HW341_4=1

**All relays off**
# sudo ./usbrelay HW341_1=0 HW341_2=0 HW341_3=0 HW341_4=0

**If for some reason the USB id changes, (ie other than XXXX:XXXX) set the USBID environment variable to the correct ID**
# sudo USBID=XXXX:XXXX ./usbrelay

For windows the code you mentioned before works fine: http://www.winddeal.net/image/tools/26484%20%2026483%20USBRelayExtLib.rar

I hope this helps you. For me it worked fine with several devices on multiple Pi's. If you have any questions, don't hesitate to ask.

twinee, 2017-06-05:
awesome, thank you

darrylb123, 2017-06-05:
Or you could try
apt-get install usbrelay

The package is available for raspian and will install the dependencies.

pvangorp, 2017-06-06:
Can you give me an update if it worked for you?

twinee, 2017-06-06:
I ended up using https://github.com/trezor/cython-hidapi, a python wrapper for hidapi, because i will be writing the rest of the control program in python and it saves me from writing my own wrapper.
I used the instructions from the readme, worked like a charm in python3 on raspberry pi. The only things i had to manually add implement were the VID&PID and the commands for the relays.
example code (adapted from try.py in trezor's git, randomly sends commands to the relays, making the board sing a little clicking song):
```
import hid
import time
import random

h = hid.device()
h.open(0x16c0,0x05df)
h.set_nonblocking(1)
now = time.time()
later = now + 10
cmd = 0x00
idx = 0x00
while time.time() < later:
    #generate random command
    idx = random.randint(0,4)
    if idx < 1:
        if random.randint(0,1) < 1:
            cmd = 0xFE # all on
        else:
            cmd = 0xFC # all off
    elif random.randint(0,1) < 1:
        cmd = 0xFF # 1 relay on
    else:
        cmd = 0xFD # 1 relay off
    #send command ([report number, command, relay_index] + padding)
    h.write([0x00, cmd, idx] + [0] * 6)
    time.sleep(0.15)
#all off
h.write([0x00, 0xFC, 0x00] + [0] * 6)
h.close()
```

 I have the 4-channel board, they all work perfectly now.

I'll send this git and cython-hidapi to sossolutions.nl because they still have the incompatible ResearchDesignLab ftdi instuction video in the product description, which cost me hours and hours of trying to find out why it didn't work and getting a replacement board.

pvangorp, 2017-06-06:
Good to hear it is working. I have send my sollution to them, but they don't do a thing with it.

twinee, 2017-06-06:
I haven't tried it, but this solution looks like it should work too for other people that would like to use python but don't want to bother with installing cython:
https://github.com/pavel-a/usb-relay-hid/blob/master/Test/test2m.py

It is an open-source fork of the DLL from the guy on ebay. they used CTypes to communicate directly with the C library. It's a little verbose but a nice interface wouldn't be hard to make.

twinee, 2017-06-06:
ha, the NO/NC labels on the board are the switched around, talk about a safety hazard. what a total piece of crap.

darrylb123, 2017-06-09:
pvangorp, 
It's so long since I set up my rPi to run my watering system that I forgot how I set it up. I am not using the package but compiled it myself. Sorry about that.
I might give the debian developer that looks after the package a poke.

closed 2018-01-27 by darrylb123

tobiz, 2019-11-23:
Can this be made to work without using sudo, if so how?
I want to be able to control a usb relay from within another program (mythtv actually) and don't/can't/musn't use sudo.

darrylb123, 2019-11-23:
Use the doco about adding udev to set the permissions when it is plugged in.

tobiz, 2019-11-24:
I tried you udev plus lots of others configs without success; result, **without** sudo gives the error:

  File "usbrelay.py", line 121, in <module>
    relay = Relay(idVendor=0x16c0, idProduct=0x05df)
  File "usbrelay.py", line 30, in __init__
    self.h.open(idVendor, idProduct)
  File "hid.pyx", line 66, in hid.device.open
IOError: open failed

**With** sudo get:

Relay initial state:  False
Current state was False, set:  True

I modified your code to see what's happening, ie it works showing swapping from one state to another.

I don't have your exact udev config at the moment but the result is the same; I can submit the results with your udev if that will help.
Any ideas as to what might be wrong welcome. 
Where's this hid.pyx file so I can see what it's doing?
Running on Ubuntu 18.04.

darrylb123, 2019-11-25:
I just set up on Fedora31 (Sorry, don't run Ubuntu) but did run into
permissions with udev.
$ ls -l /dev/usb
total 0
crw-------. 1 root root 180, 96 Nov 25 15:42 hiddev0
[dbond@zinc ~]$ usbrelay
unable to open device
: Permission denied
Device Found
  type: 16c0 05df
  path: /dev/hidraw0
  serial_number:
  Manufacturer: www.dcttech.com
  Product:      USBRelay2
  Release:      100
  Interface:    0
  Number of Relays = 2
_1=0
_2=0
I modified the configuration of udev to  to remove the bus number:
SUBSYSTEM=="usb", ATTR{idVendor}=="16c0",ATTR{idProduct}=="05df",
MODE="0666"
KERNEL=="hidraw*", ATTRS{idVendor}=="16c0", ATTRS{idProduct}=="05df",
MODE="0666"

now the permissions on the hiddev are writable by anyone
$ ls -l /dev/usb/
total 0
crw-rw-rw-. 1 root root 180, 96 Nov 25 15:54 hiddev0

$ usbrelay
Device Found
  type: 16c0 05df
  path: /dev/hidraw0
  serial_number:
  Manufacturer: www.dcttech.com
  Product:      USBRelay2
  Release:      100
  Interface:    0
  Number of Relays = 2
PSUIS_1=0
PSUIS_2=0

Don't know why the bus number was there, ancient history I suppose.
Shouldn't matter that it isn't there. It still verifies the Vendor/Product.

Try that
Darryl