darrylb123/usbrelay #16: How to get it working in javascript under Windows environment using serialport

Ryebreadizzle, 2017-05-28:
I apologize for this is not an issue but the only way I found I could reach you. I have the 2 channel usb relay module and have established a connection with it in javascript but don't know the commands to make it work. I have bought a similar usb relay module (a one channel module) that can be found here: https://goo.gl/a04gvL and have gotten it to work in a similar fashion: Establishing a serial port connection and writing to the port given the instructions. However I have found no such instructions for this device. Would you help me out?

darrylb123, 2017-05-28:
I wouldn't know where to start. The device uses the hid protocol.
Why not use the supplied windows c libraries and write a wrapper to call
from JavaScript.

Darryl

Ryebreadizzle, 2017-05-29:
Thanks for the quick response! How exactly would I go about doing that: I am fairly new to this level of programming. Is there a tutorial or such that could assist me?

Ryebread

darrylb123, 2017-05-29:
Sorry, I'm an old school Linux C programmer. Not familiar with Windows or
JavaScript programming. I suggest Google is your friend.

Ryebreadizzle, 2017-05-31:
Okay, thanks for your quick replies.

closed 2018-01-27 by darrylb123