darrylb123/usbrelay #17: Segmentation fault

tilutza, 2017-08-09:
Hi,

I am using the usbrelay pretty well. However, I have a scheduled job and after a while I have noticed that the program stop working. In the dmesg, I have a bunch of messages:

[  640.894574] usbrelay[4976]: segfault at 20 ip 00007fc5935672e0 sp 00007ffed82f1cd0 error 4 in libc-2.25.so[7fc59352f000+19d000]
[  671.907217] usbrelay[4997]: segfault at 20 ip 00007f7e3fd8c2e0 sp 00007ffc554b7d70 error 4 in libc-2.25.so[7f7e3fd54000+19d000]
[  702.895423] usbrelay[5009]: segfault at 20 ip 00007f9b1ebb42e0 sp 00007ffdc6fde390 error 4 in libc-2.25.so[7f9b1eb7c000+19d000]
[  731.060219] usbrelay[5026]: segfault at 20 ip 00007fb84c0242e0 sp 00007fff2d290ef0 error 4 in libc-2.25.so[7fb84bfec000+19d000]
[  731.370189] usbrelay[5031]: segfault at 20 ip 00007fe1ee2222e0 sp 00007fffe8029990 error 4 in libc-2.25.so[7fe1ee1ea000+19d000]
[  733.887483] usbrelay[5047]: segfault at 20 ip 00007fa78ca842e0 sp 00007ffd306279b0 error 4 in libc-2.25.so[7fa78ca4c000+19d000]
[  744.658830] usbrelay[5057]: segfault at 20 ip 00007ff23fb5d2e0 sp 00007ffe3f0c1770 error 4 in libc-2.25.so[7ff23fb25000+19d000]
[  744.926952] usbrelay[5064]: segfault at 20 ip 00007feef9ef92e0 sp 00007ffcb27ed400 error 4 in libc-2.25.so[7feef9ec1000+19d000]
[  764.895788] usbrelay[5087]: segfault at 20 ip 00007f40fc1d92e0 sp 00007ffe376b3750 error 4 in libc-2.25.so[7f40fc1a1000+19d000]
[  795.896736] usbrelay[5098]: segfault at 20 ip 00007f30316ac2e0 sp 00007ffd4c006500 error 4 in libc-2.25.so[7f3031674000+19d000]
[  826.901807] usbrelay[5136]: segfault at 20 ip 00007fc0dafc52e0 sp 00007ffdf7400e30 error 4 in libc-2.25.so[7fc0daf8d000+19d000]
[  857.891152] usbrelay[5147]: segfault at 20 ip 00007f8836b9a2e0 sp 00007ffe1f527820 error 4 in libc-2.25.so[7f8836b62000+19d000]
[  888.893886] usbrelay[5165]: segfault at 20 ip 00007f06e26342e0 sp 00007ffd7b7e2790 error 4 in libc-2.25.so[7f06e25fc000+19d000]
[  919.896832] usbrelay[5182]: segfault at 20 ip 00007ff071be52e0 sp 00007ffc587fbd20 error 4 in libc-2.25.so[7ff071bad000+19d000]
[  950.899590] usbrelay[5200]: segfault at 20 ip 00007fd7352ec2e0 sp 00007ffe405c1d40 error 4 in libc-2.25.so[7fd7352b4000+19d000]

If I run the usbrelay with strace, I see this:

8, TFD_TIMER_ABSTIME, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=76435, tv_nsec=590560000}}, NULL) = 0
ioctl(9, USBDEVFS_SUBMITURB, 0x798a68b100) = 0
poll([{fd=6, events=POLLIN}, {fd=8, events=POLLIN}, {fd=9, events=POLLOUT}], 3, 60000) = 1 ([{fd=9, revents=POLLOUT}])
ioctl(9, USBDEVFS_REAPURBNDELAY, 0x7fffe8206960) = 0
timerfd_settime(8, 0, {it_interval={tv_sec=0, tv_nsec=0}, it_value={tv_sec=0, tv_nsec=0}}, NULL) = 0
ioctl(9, USBDEVFS_REAPURBNDELAY, 0x7fffe8206960) = -1 EAGAIN (Resource temporarily unavailable)
write(7, "\1", 1)                       = 1
close(9)                                = 0
open("/sys/bus/usb/devices/1-1/bConfigurationValue", O_RDONLY) = 9
read(9, "1\n", 5)                       = 2
close(9)                                = 0
open("/sys/bus/usb/devices/1-1/bConfigurationValue", O_RDONLY) = 9
read(9, "1\n", 5)                       = 2
close(9)                                = 0
open("/sys/bus/usb/devices/usb1/bConfigurationValue", O_RDONLY) = 9
read(9, "1\n", 5)                       = 2
close(9)                                = 0
open("/sys/bus/usb/devices/usb1/bConfigurationValue", O_RDONLY) = 9
read(9, "1\n", 5)                       = 2
close(9)                                = 0
write(2, "Device Found\n  type: 16c0 05df\n "..., 75Device Found
  type: 16c0 05df
  path: 0001:0007:00
  serial_number: (null)) = 75
write(2, "\n", 1
)                       = 1
write(2, "  Manufacturer: (null)\n", 23  Manufacturer: (null)
) = 23
write(2, "  Product:      (null)\n", 23  Product:      (null)
) = 23
write(2, "  Release:      100\n", 20  Release:      100
)   = 20
write(2, "  Interface:    0\n", 18  Interface:    0
)     = 18
--- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x20} ---
+++ killed by SIGSEGV (core dumped) +++
[1]    17365 segmentation fault  sudo strace /home/home-assistant/automation/usbrelay

I noticed a few things:
- the output of the program is like this:
Device Found
  type: 16c0 05df
  path: 0001:0007:00

The path is not displayed correctly. However I have noticed that the device is mounted in /dev/hidraw3

If i unplug the device and plug it again, everything works. 

I am using  ARCH LINUX (Linux 4.12.4-1-ARCH #1) and I have used gcc 7.1 to compile the program.

Do you have any idea why this behavior is only after a matter of time ?

tilutza, 2017-08-09:
Actually I think it's a problem in the HID driver (maybe caused by the usbrelay software). If I list the usb particularities with 
lsusb -v -d 16c0:05df 

I see at the end

.....
          wDescriptorLength      22
          Report Descriptor: (length is -1)
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval              20
can't get debug descriptor: Resource temporarily unavailable
cannot read device status, Resource temporarily unavailable (11)

darrylb123, 2017-08-13:
Except that lsusb is not using the HID driver. I'd say there is a hardware
fault. Does it do the same on any USB port?
When you unplug and replug, does the fault show up with lsusb or only after
a while?
This is what mine looks like with lsusb -v -d 16c0:05df
...
         wDescriptorLength      22
         Report Descriptors:
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval             100
Device Status:     0x0000
  (Bus Powered)

closed 2018-01-27 by darrylb123