darrylb123/usbrelay #18: How to use in script if two one channel usb relay with same ID

combat007, 2018-01-25:
Your program works great on one channel relay
But I am stuck with the problem that I have two 1 channel relays attached to two usb ports of my laptop.
Following is the output of the command ./usbrelay

Device Found
  type: 16c0 05df
  path: /dev/hidraw2
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay1
  Release:      100
  Interface:    0
  Number of Relays = 1
959BI_1=0

Device Found
  type: 16c0 05df
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay1
  Release:      100
  Interface:    0
  Number of Relays = 1
959BI_1=0

When I provide the command "./usbrelay 959BI_1=1" it works on both the relays simultaneously, but I want that only one must work at a time. Wht both channels device showing same ID?
How can I differentiate between those two relay channel in the command
./usbrelay <ID>
??

darrylb123, 2018-01-25:
Well that's a problem. I can't think of any way to discriminate other than
buying another. They should have unique id.
Try having the supplier replace it.

combat007, 2018-01-26:
ID is provided as a parameter to ./usbrelay. Can't we provide the device path instead??
I think it will solve the problem if we can provide the device path.

combat007, 2018-01-26:
well I followed this link and got the solution
https://www.quora.com/How-do-I-enable-and-disable-USB-port-using-terminal-in-Ubuntu

then I wrote the script "mycmd"

#  ---------Bash Script start here -------  #
if [ $1 == "small" ]
then
	echo '2-1' | tee /sys/bus/usb/drivers/usb/unbind
	./usbrelay 959BI_1=1;./usbrelay 959BI_1=0
	echo '2-1' | tee /sys/bus/usb/drivers/usb/bind
fi

if [ $1 == "big" ]
then
	echo '2-3' | tee /sys/bus/usb/drivers/usb/unbind
	./usbrelay 959BI_1=1;./usbrelay 959BI_1=0
	echo '2-3' | tee /sys/bus/usb/drivers/usb/bind
fi

# ---------Bash Script end here -------  #

Now this helped me out to start two machines one by one
./mycmd small 
./mycmd big

the rest of the job was done by your utility and it worked like a charm. 
Kindly share this solution to those who are seeking it like I was.

Regards

darrylb123, 2018-01-26:
Thanks. It really should not be necessary. The ID should be unique.
I'd be careful as the USB bus instantiates the device addresses non
predictably. Try a few power cycled to make sure the devices come up as
expected.

danbayona, 2018-03-14:
I'm having the same problem -- except I can't use the solution that combat007 provides. I can't turn any of them off as I have to potentially trigger channels simulataneously... Would it be possible to somehow provide device path + ID to the usbrelay command?

I currently have 3 boards and they are all given the same buf IDs (6QMBS_1 - 6QMBS_8). I feel it is highly unlikely that getting a replacement will enumerate unique IDs, but if it really is the only chance I guess i'll try!

darrylb123, 2018-03-14:
Well that's very slack of the vendors. The trouble with device ID is that
it will depend how the USB devices are enumerated could result in
unpredictable device IDs. Not a good situation when connected to real
equipment.

I will have a look at it.

danbayona, 2018-03-14:
Would it change anything if I use udev rules so that the USB relay devices are enumerated statically? I currently set rules for other USB devices so they can be referred to by a non-dynamic path.

darrylb123, 2018-03-14:
That would probably work.

danbayona, 2018-03-14:
The problem is -- even with different /dev/ paths the channel ID's are still identical. Am I missing something? When you refer to 'device ID' do you mean the ID '16c0:05df'?

darrylb123, 2018-03-15:
that's the problem. The manufacturer isn't randomising the HID ID and the
device type is fixed. The only way of doing it is to point to the USB port.
Should at least complain to the vendor about randomising the HID ID. Their
software won't work either.

darrylb123, 2018-03-15:
Found the author of replacement firmware for the board. It is possible to
change the serial. I'll look at the windows code to see if it can do it.
Otherwise, I'll see if I can add the functionality to usbrelay.

danbayona, 2018-03-15:
Wow thank you so much for your efforts! fyi I am running this on linux Ubuntu 16.04

darrylb123, 2018-03-16:
Hold your thanks until I get it to work :)

darrylb123, 2018-03-17:
Ok, please test the new version. I have added the capability to change the serial. I have updated the documentation to show how to do it.
Worked fine on my test board. plug in one of the devices with the duplicate serial and change it.

darrylb123, 2018-03-21:
Can you test my changes. I have updated the doco to show how to change the
serial. You will need to pull the latest version and recompile.

danbayona, 2018-03-21:
My apologies! I was on travelling/on vacation until today. Will give it a test run today and let you know.

Sent via the Samsung Galaxy Note5, an AT&T 4G LTE smartphone

  relay with same ID (#18)

danbayona, 2018-03-21:
Hey! The IDs for each channel are now able to be enumerated uniquely and will work for my purposes, thank you so much for your time!

darrylb123, 2018-03-22:
Tested by danbayona, tested by myself on 2 way and 8 way relay.
$ ./usbrelay 
Device Found
  type: 16c0 05df
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
86UXU_1=0
86UXU_2=0
86UXU_3=0
86UXU_4=1
86UXU_5=0
86UXU_6=0
86UXU_7=0
86UXU_8=0

$ ./usbrelay 86UXU_0=86UXV
Orig: 86UXU, Serial: 86UXU, Relay: 0 State: 0
Device Found
  type: 16c0 05df
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
Serial: 86UXU, Relay: 0 State: 0 
1 HID Serial: 86UXU 
Serial: 86UXU, Relay: 0 State: 0 --- Not Found
 $ ./usbrelay 
Device Found
  type: 16c0 05df
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
86UXV_1=0
86UXV_2=0
86UXV_3=0
86UXV_4=1
86UXV_5=0
86UXV_6=0
86UXV_7=0
86UXV_8=0

$ ./usbrelay 86UXV_0=86UXU
Orig: 86UXV, Serial: 86UXV, Relay: 0 State: 0
Device Found
  type: 16c0 05df
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
Serial: 86UXV, Relay: 0 State: 0 
1 HID Serial: 86UXV 
Serial: 86UXV, Relay: 0 State: 0 --- Not Found
 $ ./usbrelay 
Device Found
  type: 16c0 05df
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
86UXU_1=0
86UXU_2=0
86UXU_3=0
86UXU_4=1
86UXU_5=0
86UXU_6=0
86UXU_7=0
86UXU_8=0

closed 2018-03-22 by darrylb123

ImXron, 2018-03-25:
Hi,

I was working with Dan on the issue above. I built your new changes using Docker so I didn't have to install any dependencies locally. Would you be interested in adding the Dockerfile, a build script and an updated Readme to the project? I have branch ready to push. This way, anyone can build from source once they install Docker.

Thanks for the help!

darrylb123, 2018-03-25:
Sure