darrylb123/usbrelay #22: link to purchase this

nx-99, 2018-06-30:
Hi, it's possible post  link to purchase this, thank.

pauledd, 2018-07-23:
This one worked for me:
https://de.aliexpress.com/item/4-channel-DC-12V-computer-USB-control-switch-drive-relay-module-PC-intelligent-controller-4-way/32806651106.html?spm=a2g0x.10010108.1000016.1.3efb4f425q4XpD&isOrigTitle=true

nx-99, 2018-07-23:
Thank you very much
Grazie mille,

nx-99, 2018-07-24:
Thank darrylb123, I am interesting at the 1 Channel 5V, it work fine whit Ubuntu 16.04 ?

bres55, 2018-08-18:
Thanks for swift reply, and suggestions. Not sure I want to go that far, probably easier to buy the 4 or maybe 8 relay versions. I have tried using the program supplied by the seller, CommandApp_USBRelay.exe, and tried permutations like, CommandApp_USBRelay HURTM=HURT1, but no joy so far. I've been in touch with seller, maybe they will come up with a solution. Thanks again. Bres55

darrylb123, 2018-08-18:
That application does not support changing the serial. Booting Linux on USB
stick works very well and does not affect your installed windows

bres55, 2018-08-18:
OK. saves me trying to achieve the impossible! :-) 
I'm hoping to use the relays in my solar-powerwall set up, don't feel I have the enthusiasm to get side tracked just now. I might get a raspberry pi, in the future to make all my settings visible in a browser. So I might have a go then. Toying with battery monitoring, balancing and remote switching is enough to keep me busy for now. Thanks again.