darrylb123/usbrelay #23: change initial state?

pauledd, 2018-07-23:
Hi

I am not very familiar with the internals of the relay hardware but I am just curious about if
it would be possible to modify the relay somehow so that all relays switch to ON on power-up
(as default state) even without USB connected, just 12V powering. Is that somehow possible?

darrylb123, 2018-07-23:
The relay contacts are double throw. They have a common with a normally
open and normally closed. So you would connect to the common and normally
closed to achieve what you want.

darrylb123, 2018-07-23:
> That wont work for my purpose... I want to power my Raspberry Pi through
> the relay, and turn
> it off as the Raspberry Pi issues a shutdown command. Then to power it on
> again I want to
> tip a pushbutton that powers up the relay (just like starting a car) that
> closes the power switch and stays closed so that the Raspberry Pi can boot
> up again. So I need the relay to be open if not powered, and closed if
> powered up (without issuing a usb close command).
>
> [image: 20180723_115751]
> <https://user-images.githubusercontent.com/12088907/43070414-110678a0-8e70-11e8-88ee-dc27cc990bd7.jpg>
>
> I currently have a startup script that closes the relay on boot, but I
> have to keep-pressed the switch
> until the bootscript gets executed (20-30secs...) but I could live with
> that :)
>
> I originally had the pushbutton and a non-USB relay with low-active
> characteristics that did the job...
> B

Perhaps an additional timer relay that powers the pi until it can take
control.