darrylb123/usbrelay #24: Are FTDI based relays supported?

xj25vm, 2018-07-30:
I just got an 8 channel FTDI based usb relay and I was wondering if usbrelay can work with it?  It creates a /dev/ttyUSB0 and the relevant info is:

lsusb -v -d 0403:6001

Bus 004 Device 020: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0403 Future Technology Devices International, Ltd
  idProduct          0x6001 FT232 Serial (UART) IC
  bcdDevice            6.00
  iManufacturer           1 FTDI
  iProduct                2 FT245R USB FIFO
  iSerial                 3 A600OJ7L
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0020
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower               90mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           2
      bInterfaceClass       255 Vendor Specific Class
      bInterfaceSubClass    255 Vendor Specific Subclass
      bInterfaceProtocol    255 Vendor Specific Protocol
      iInterface              2 FT245R USB FIFO
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x02  EP 2 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               0
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)

darrylb123, 2018-07-30:
No, usbrelay is for HID class USB.

From what I understand, these relays just need a certain string sent to the
USB serial port /dev/ttyUSB0.
I saw one the other day which documented the strings in the ad:
LCUS-2 Dual Channel USB Relay Module USB Intelligent Control Switch
https://banggood.app.link/jcjREeM3YO

In fact, the strings look the same as the esp8266 based wireless relay that
I got working with arduino code in the esp8266.

xj25vm, 2018-07-31:
Thank you for the reply Darryl. In case anybody else finds it useful, I stumbled over another project which can work with FTDI relays - https://github.com/ondrej1024/crelay

On a different note, I use usbrelay for other projects which use HID chips - and find it extremely useful. Thank you for working on this and sharing it with the rest of us!

darrylb123, 2018-07-31:
Cool, what are the other projects? Do you have to modify the code or are
you using it as an example? I'd be happy to incorporate useful devices
where it made sense.

xj25vm, 2018-07-31:
I mainly use it with 1 or 2 channel usb relays at several sites to reboot ADSL modems on troublesome lines - which get disconnected from the internet and stay like that from time to time. I use Monit on the main server (to which the usb relay is connected) - which detects when the internet connection is lost and sends an 'off' and then 'on' command to the usb relay. I find that functionality extremely useful - specially on sites with nobody around out of office hours!