darrylb123/usbrelay #25: unable to change serial number

rickyhp, 2018-10-06:
Hi,
I'm trying to change device serial number using
sudo usbrelay OLD_0=NEW but it does not get changed. Can please advise?

sudo usbrelay HURTM_0=ABCDE
Orig: HURTM, Serial: HURTM, Relay: 0 State: fd
Device Found
  type: 16c0 05df
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay2
  Release:      100
  Interface:    0
  Number of Relays = 2
Serial: HURTM, Relay: 0 State: fd 
1 HID Serial: HURTM Serial: HURTM, Relay: 0 State: fd

Serial: HURTM, Relay: 0 State: fd --- Found

darrylb123, 2018-10-07:
What does it output when you run usbrelay a second time.
Did you compile from source or are you running the Debian package. The
ability to change the set the serial is a recent feature.
Please compile the latest version.

darrylb123, 2018-10-07:
> Hi Darryl,
> When I run second time, it shows same serial number only i.e. HURTM in my
> case.
> I run from debian package. Thanks.

The Debian package is probably the old version Build it from the code on
github and try again.

rickyhp, 2018-10-07:
Hi Darryl,
Yes It works after I rebuild from latest code. Thanks.

closed 2019-03-22 by darrylb123

marcpuca, 2019-12-30:
Hi Darryl, 
I am facing the same issue on ubuntu 18.04, but I do not have a clue on how to build from source, any tips would be appreciated
Thanks

darrylb123, 2019-12-30:
Try getting a clue from the doco. Tell me what you don't understand and
I'll try to help.

marcpuca, 2019-12-30:
I found another thread where you pointed the way!
Sorry for bothering , Thanks  and a Happy New Year

darrylb123, 2019-12-30:
What information was in the thread and not in the doco? I'll try to fix it.

marcpuca, 2019-12-30:
Line by line instructions for compiling e.g.

"I just tested using a Ubuntu 14.04 x86_64 Virtual machine
I did this:
sudo apt-get install libhidapi-dev libhidapi-hidraw0 git
git clone https://git.com/darrylb123/usbrelay
cd usbrelay
gcc -o usbrelay usbrelay.c -lhidapi-hidraw0
sudo ./usbrelay
sudo ./usbrelay L2FGQ_1=1

and it worked exactly like fedora
apt-cache showpkg libhidapi-dev
shows it is 0.8rc1"

I think I actually used make command rather than compiling with gcc, not
sure if this is present in the doco, if it is then apologies.
Thanks

darrylb123, 2019-12-30:
Yes, make command is best. Did you follow the python build in the doco too?

darrylb123, 2019-12-31:
I've added line by line build instructions. Can you have a look and see if
they make sense?

Darryl

marcpuca, 2019-12-31:
No python, used make, then had to copy libusbrelay.so to usr/local/lib to
get executable to run

Thanks

darrylb123, 2019-12-31:
Makefile installs the library into /use/lib(64)

I've updated the docs to describe it all
Run:
make
sudo make install

marcpuca, 2019-12-31:
Excellent thanks