darrylb123/usbrelay #26: After modifying the name of the board, but the name becomes empty.

jingrongx, 2018-11-20:
After modifying the name of the board, the name becomes empty
Then I can't change it back.

```

root@acrn-io-ubuntu-xujr:/usb-relay/usbrelay# ls -l
总用量 408
-rw-r--r--  1 root root    412 11月  6 16:04 build.sh
-rw-r--r--  1 root root    127 11月  6 16:04 Dockerfile
drwxr-xr-x 16 root root   4096 11月  6 16:07 hidapi
-rw-r--r--  1 root root  18026 11月  6 16:04 LICENSE.md
-rw-r--r--  1 root root    298 11月  6 16:04 Makefile
-rw-r--r--  1 root root   6604 11月  6 16:04 README.md
-rwxr-xr-x  1 root root  18096 11月  6 16:05 usbrelay
-rw-r--r--  1 root root   6901 11月  6 16:04 usbrelay.c
-rw-r--r--  1 root root   1116 11月  6 16:04 usbrelay.h
-rw-r--r--  1 root root 337602 11月  6 16:04 usbrelay.jpg
root@acrn-io-ubuntu-xujr:/usb-relay/usbrelay# ./usbrelay 2>/dev/null
_1=0
_2=0
_3=0
_4=0
_5=0
_6=0
_7=0
_8=0
root@acrn-io-ubuntu-xujr:/usb-relay/usbrelay# ./usbrelay _0=HW554
Orig: _0, Serial: 0, Relay: 0 State: 0
Device Found
  type: 16c0 05df
  path: 0001:0009:00
  serial_number: (null)
  Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
Serial: 0, Relay: 0 State: 0

Serial: 0, Relay: 0 State: 0 --- Not Found
root@acrn-io-ubuntu-xujr:/usb-relay/usbrelay# ./usbrelay 2>/dev/null
_1=0
_2=0
_3=0
_4=0
_5=0
_6=0
_7=0
_8=0
root@acrn-io-ubuntu-xujr:/usb-relay/usbrelay#

```

darrylb123, 2018-11-20:
What did you change it to? Was it a unicode string. I've not tested for
anything but ASCII. I'll have a look and see what can be done.

jingrongx, 2018-11-20:
Thank you for your reply!
There was one time that I change the board's name, but it's failed.

jingrongx, 2018-11-20:
Found it, it was this order.

`./usbrelay MRB12_0=MRB1O`

darrylb123, 2018-11-20:
> Can you copy the command that you used? I can't see how it would accept an
> empty serial?
> $ sudo ./usbrelay HW554_0=""
> Orig: HW554, Serial: HW554, Relay: 0 State: 0
> Device Found
>   type: 16c0 05df
>   path: /dev/hidraw4
>   serial_number:
>   Manufacturer: www.dcttech.com
>   Product:      USBRelay2
>   Release:      100
>   Interface:    0
>   Number of Relays = 2
> Serial: HW554, Relay: 0 State: 0
> 1 HID Serial: HW554
>
>  New serial can't be empty!
>
> Serial: HW554, Relay: 0 State: 0 --- Not Found

Can you edit the code and comment out 2 of lines and recompile. ( comments
are the // at the beginning of the line )

for (i=1;i<argc;i++) {
         fprintf(stderr,"Serial: %s, Relay: %d State: %x
\n",relays[i].this_serial,relays[i].relay_num,relays[i].state);
//          if (!strcmp(relays[i].this_serial, (const char *) buf)) {
              /* This one is now commented out */
            fprintf(stderr,"%d HID Serial: %s ", i, buf);
    if (relays[i].relay_num == 0 ) {
                if ( ! relays[i].new_serial[0] ) {
                    fprintf(stderr,"\n \n New serial can't be empty!\n");
                } else {
                    set_serial(handle,relays[i].new_serial);
                }
            } else {
               fprintf(stderr,"Serial: %s, Relay: %d State:
%x\n",relays[i].this_serial,relays[i].relay_num,relays[i].state);
               if
(operate_relay(handle,relays[i].relay_num,relays[i].state) < 0 )
   exit_code++;
               relays[i].found = 1;
           }
//          }                      /* and this one */
      }

The code will then match any plugged in relay device, so only have one
relay board plugged in.
Use your command
./usbrelay _0=HW554
Make sure you remove the comments and recompile before using usbrelay after
fixing the serial.

I would very much like to know what command you used to I can make sure
nobody else is caught.

darrylb123, 2018-11-20:
Well I don't know, looks fine to me?

sudo ./usbrelay MRB12_0=MRB1O
Orig: MRB12, Serial: MRB12, Relay: 0 State: 0
Device Found
  type: 16c0 05df
  path: /dev/hidraw4
  serial_number:
  Manufacturer: www.dcttech.com
  Product:      USBRelay2
  Release:      100
  Interface:    0
  Number of Relays = 2
Serial: MRB12, Relay: 0 State: 0
1 HID Serial: MRB12
Serial: MRB12, Relay: 0 State: 0 --- Not Found
[dbond@bashful usbrelay]$ sudo ./usbrelay
Device Found
  type: 16c0 05df
  path: /dev/hidraw4
  serial_number:
  Manufacturer: www.dcttech.com
  Product:      USBRelay2
  Release:      100
  Interface:    0
  Number of Relays = 2
MRB1O_1=0
MRB1O_2=0

jingrongx, 2018-11-20:
Very strange, I just want to give up, but I typed a character "x" , and then the serial name recovered to HW554 !

```
root@acrn-io-ubuntu-xujr:/usb-relay/usbrelay# sudo ./usbrelay 2>/dev/null
_1=0
_2=0
_3=0
_4=0
_5=0
_6=0
_7=0
_8=0

root@acrn-io-ubuntu-xujr:/usb-relay/usbrelay# ./usbrelay x_0=HW554
Orig: x, Serial: x, Relay: 0 State: 0
Device Found
  type: 16c0 05df
  path: /dev/hidraw0
  serial_number:
  Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
Serial: x, Relay: 0 State: 0
1 HID Serial:
Serial: x, Relay: 0 State: 0 --- Not Found

root@acrn-io-ubuntu-xujr:/usb-relay/usbrelay# sudo ./usbrelay 2>/dev/null
HW554_1=0
HW554_2=0
HW554_3=0
HW554_4=0
HW554_5=0
HW554_6=0
HW554_7=0
HW554_8=0
```

darrylb123, 2018-11-20:
Ha, sorry,
Yes, it needed something

jingrongx, 2019-03-15:
the board's name becomes garbled characters

root@test-ssd:~/usbrelay# ./usbrelay
Device Found
  type: 16c0 05df
  path: /dev/hidraw6
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
�#}��_1=0
�#}��_2=0
�#}��_3=0
�#}��_4=0
�#}��_5=0
�#}��_6=0
�#}��_7=0
�#}��_8=0

darrylb123, 2019-03-15:
Hey, I noticed that the other day.
Run it as root or use sudo and it will work.

mudplugga, 2019-07-27:
Hi, I've been using my board without any problems.....untill the other day...
It is detected, but the ports are garbled  ��Z54_1=0. What was the solution to this problem,? I've tried following the previous posts but no success...
.terry@latitude:~/Projects/usbrelay$ usbrelay
Device Found
  type: 16c0 05df
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
��Z54_1=0
��Z54_2=0
��Z54_3=0
��Z54_4=0
��Z54_5=0
��Z54_6=0
��Z54_7=0
��Z54_8=0

cheers
Terry

darrylb123, 2019-07-28:
You need to use root privileges. Use Sudo.

closed 2019-12-01 by darrylb123