darrylb123/usbrelay #32: Raspberry pi 4 Enumerate

ilikemyfto, 2019-08-02:
Hi Darryl

Thank you for the use of your usbrelay2 code for the past couple of years. In 2017 I purchased a wonderful coffee machine that unfortunately suffers from a gashly EU ruling that forces the machine to switch off after 30 minutes of inactivity. Fair enough but to get the machine to the peak running state it needs 45 minutes at least to warm up. 

I discovered that the on/off switch just needed a pulse within 30 minutes to keep the machine running and the usbrelay for ebay was the beginning of the solution. I found your code on GitHub and a friend helped me to get it running on an early Raspberry pi B. I used crontab to schedule the events and it would switch the machine on in the morning and switch it off after lunch and provided a pulse every 20 minutes to keep the timer running.

It has run well (headless) for the 2 years but now connecting to it is difficult. The ethernet port now seems to have activity and so I purchased a new Raspberry PI 4. Unfortunately, your code no longer seems to compile and my friend is at a loss as to the error.

usbrelay.c:(.text+0x36c): undefined reference to `enumerate_relay_boards'
/usr/bin/ld: usbrelay.c:(.text+0x42c): undefined reference to `find_board'
/usr/bin/ld: usbrelay.c:(.text+0x4f8): undefined reference to `set_serial'
/usr/bin/ld: usbrelay.c:(.text+0x5dc): undefined reference to `operate_relay'
collect2: error: ld returned 1 exit status

Attached is a copy of my installation procedure and also a copy of the compile run. My knowledge of C++ is limited together with my experience of the cmd line. Would you be able to look at the attached file and hopefully give some pointers to the solution.

Many thanks in advance.

KR

Roy
[Raspberry Pi - USBRELAY compile commands.docx](https://github.com/darrylb123/usbrelay/files/3462366/Raspberry.Pi.-.USBRELAY.compile.commands.docx)
[Run gcc -v -o.docx](https://github.com/darrylb123/usbrelay/files/3462369/Run.gcc.-v.-o.docx)

darrylb123, 2019-08-03:
If you run make in usbrelay directory you should be ok.
ie:
cd usbrelay
make

ilikemyfto, 2019-08-04:
I also added "sudo make install" and it burst into life.
Many thanks Darryl for your help.
KR
Roy

closed 2019-08-04 by ilikemyfto

darrylb123, 2019-08-05:
I've added some better doco for building the software.

tirfil, 2019-10-01:
Hello,
I have met the same issue on Ubuntu.
The Makefile contains an error.
Change
usbrelay: usbrelay.c libusbrelay.h libusbrelay.so
	$(CC) $(CPPFLAGS) $(CFLAGS) $(LDFLAGS) -L./ -lusbrelay -o $@ $<
With
usbrelay: usbrelay.c libusbrelay.h libusbrelay.so
        $(CC) $(CPPFLAGS) $(CFLAGS) $< -lusbrelay -L./ $(LDFLAGS) -o $@

In gcc command line, source code (.c) must always appear before shared libraries (-l) 

Regards,
Philippe

darrylb123, 2019-10-04:
What version of gcc do you use with your version of Ubuntu?
My Fedora version builds it just fine.
cc -v
Using built-in specs.
COLLECT_GCC=cc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/9/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-redhat-linux
Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,fortran,objc,obj-c++,ada,go,d,lto --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --with-isl --enable-offload-targets=nvptx-none --without-cuda-driver --enable-gnu-indirect-function --enable-cet --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux
Thread model: posix
gcc version 9.2.1 20190827 (Red Hat 9.2.1-1) (GCC) 

so my Makefile output looks like this:
cc -shared -fPIC  -O2 -Wall -lhidapi-hidraw -L /usr/lib64 -Wl,-rpath /usr/lib64 -o libusbrelay.so libusbrelay.c
cc  -O2 -Wall -lhidapi-hidraw -L /usr/lib64 -Wl,-rpath /usr/lib64 -L./ -lusbrelay -o usbrelay usbrelay.c

darrylb123, 2019-10-04:
I tested your change on Fedora and raspian (gcc version 4.9.2 (Raspbian 4.9.2-10+deb8u2)), works for me. Pushed your change, please test for me.