darrylb123/usbrelay #33: 2 modules with the same id product and Id vendor in the same linux system

ILENIMARIUS, 2019-10-31:
Hello,
I am using a ubuntu system 16.04  with 2 modules usb-relay connected and i want to create linux usb rule to differentiate them from each other.

The python code example lets you create and object having only 2 arguments ( id vendor and id product) which is the same for both.
I was not able to find the serial number and i have seen that this kinds of modules have a special serial number but i do not know how to link this to the linux usb rules.
Could you help me please?

darrylb123, 2019-10-31:
Use the usbrelay command to get the serial. I'll have to pass your request
to the person that wrote the python code.

ILENIMARIUS, 2019-10-31:
There is no serial number when i run the command.

We have to add  a device attribute (ATTR) like serial number or something unique for each device in to /etc/udev/rules.d/50-dct-tech-usb-relay-2.rules.

EX:
SUBSYSTEM=="usb", ATTR{idVendor}=="16c0",ATTR{idProduct}=="05df", MODE="0666"
KERNEL=="hidraw*", ATTRS{busnum}=="1", ATTRS{idVendor}=="16c0", ATTRS{idProduct}=="05df", MODE="0666"

For arduino or other usb connected devices the best options is to run a linux command "_udevadm_xxx etc to check all the attributes of a usb device" and add the attribute to the ubuntu dev rules.

How can i solve this problem with the identical modules?
TX!

darrylb123, 2019-11-01:
The vendor and product is only used for udev to set the permissions on the device so that a normal user can operate the relay. It is not used to identify which relay. Each relay module needs a unique serial. See the examples in the documentation. eg:
$ sudo usbrelay
Device Found
  type: 16c0 05df
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay2
  Release:      100
  Interface:    0
PSUIS_1=1
PSUIS_2=0

Note that in this case the serial is PSUIS with 2 relays. The usbrelay command can also change the serial in the event that the serial is not unique.
I do not bother using udev to set the permissions for my applications, I use sudo to run the usbrelay command as root.

I couldn't program in Python so save myself, but I see that the serial is the first argument passed to the board_control method. From the example, which iterates over all the boards connected. I have confirmed that this test script works with multiple boards.
usbrelay_py.board_control(board[0],relay,0)

darrylb123, 2019-11-01:
This python code operates my test board:

import usbrelay_py
import time

count = usbrelay_py.board_count()
print("Count: ",count)

boards = usbrelay_py.board_details()
print("Boards: ",boards)

result = usbrelay_py.board_control("PSUIS",1,1)

The resulting output looks like this and relay 2 operates:
$ python usbrtest.py
Count:  1
Boards:  (('PSUIS', 2, 0),)

ILENIMARIUS, 2019-11-12:
Hello again,
Thank you for the replies!
I have installed the usbrelay library:
"sudo apt install usbrelay"
and then created a python instance for every module with subprocess calls like :  "usbrelay", "VD0LE_1=1"
Basically one python instance using subprocess calls sends 0 and 1 to a module with one relay and another .py instance sends calls for a two module  relays.
In the first .py instance there is a loop which sends commands  VD0LE_1=1 every 5 seconds to cclose the relay and sometimes randomly one relay at a time is activated from the other board with 2 relays. This behaviour is like something is wrong with some shared memory or something.
Every board has different code for activating the relays (ex: VLTFT_1) only the rest attrs are identical.
Where is the problem?

darrylb123, 2019-11-13:
Why don't you use the python module rather than use the shell command as a
subprocess?
The usbrelay command is used in shell scripts and for operating the relays
from the command line.

Have a look at the test.py file. This test program will operate all relays
connected to your system.
My previous email example operates one named relay from python, note it
does not use a subprocess.

If you want to use python, you really should use the python module as I
have no idea what races can occur when using subprocesses.

darrylb123, 2019-11-13:
I just realised you are not building the code here on github. I don't think
the usbrelay command installed by apt install usbrelay has the python
interface.
Please follow the instructions and build the code by cloning the github
repository.

ILENIMARIUS, 2019-11-13:
Hello,
I have followed your suggestion but unfortunately i am stuck.
When trying to run the test.py code the imported usbrelay_py modules throws a hid error.
[Traceback (most recent call last):
  File "test.py", line 1, in <module>
    import usbrelay_py
ImportError: /usr/lib/x86_64-linux-gnu/libusbrelay.so: undefined symbol: hid_error
]
I have no more ideas what to do next...

darrylb123, 2019-11-13:
Do you have the hidapi packages installed as per the documentation?

ILENIMARIUS, 2019-11-13:
Hello,
I have installed using this:
sudo apt-get install libhidapi-dev

darrylb123, 2019-11-13:
OK, but you need to install libhidapi too. I'm surprised it didn't do it
automatically. The dev package allows you to build the software the other
has the actual libraries.

ILENIMARIUS, 2019-11-14:
Hello Darryl,
I have already installed the libhidapi:

_$sudo apt-get install libhidapi-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libhidapi-dev is already the newest version (0.8.0~rc1+git20140818.d17db57+dfsg-1)._

_$sudo apt-get install libhidapi-hidraw0
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libhidapi-hidraw0 is already the newest version (0.8.0~rc1+git20140818.d17db57+dfsg-1)._

I have build everything couple of times but i still have the same error when i follow the readme steps:

_#sudo python3 test.py 
Traceback (most recent call last):
  File "test.py", line 1, in <module>
    import usbrelay_py
ImportError: /usr/lib/aarch64-linux-gnu/libusbrelay.so: undefined symbol: hid_error_

What can be different ?

darrylb123, 2019-11-14:
Ok, can you post the output when you do a complete build and install, ie
make clean
make
sudo make install

I'm not familiar with Ubuntu but that might give me a hint.

darrylb123, 2019-11-14:
Also, post the output of the python build.

##Build libusbrelay_py.so
$ make python
##Install to global python
$ sudo make install_py

ILENIMARIUS, 2019-11-14:
Hello Darryl,

Please find the output of the steps:

_root@tegra-ubuntu:~/test/usbrelay-master# make
cc -shared -fPIC  -O2 -Wall -lhidapi-hidraw -L /usr/lib/aarch64-linux-gnu -Wl,-rpath /usr/lib/aarch64-linux-gnu -o libusbrelay.so libusbrelay.c
cc  -O2 -Wall usbrelay.c -lusbrelay -L./ -lhidapi-hidraw -L /usr/lib/aarch64-linux-gnu -Wl,-rpath /usr/lib/aarch64-linux-gnu -o usbrelay
root@tegra-ubuntu:~/test/usbrelay-master# sudo make install
install -d /usr/lib/aarch64-linux-gnu
install -m 0755 libusbrelay.so /usr/lib/aarch64-linux-gnu
install -d /usr/bin
install -m 0755 usbrelay /usr/bin
root@tegra-ubuntu:~/test/usbrelay-master# make python
cc -shared -fPIC -I/usr/include/python3.5m  -O2 -Wall -lhidapi-hidraw -L /usr/lib/aarch64-linux-gnu -Wl,-rpath /usr/lib/aarch64-linux-gnu -L./ -lusbrelay -o libusbrelay_py.so libusbrelay_py.c
python3 setup.py build
running build
running build_ext
building 'usbrelay_py' extension
creating build
creating build/temp.linux-aarch64-3.5
aarch64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.5m -c libusbrelay_py.c -o build/temp.linux-aarch64-3.5/libusbrelay_py.o
creating build/lib.linux-aarch64-3.5
aarch64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 build/temp.linux-aarch64-3.5/libusbrelay_py.o -L./ -L/usr/lib/x86_64-linux-gnu -L/usr/lib/arm-linux-gnueabihf -lusbrelay -o build/lib.linux-aarch64-3.5/usbrelay_py.cpython-35m-aarch64-linux-gnu.so
root@tegra-ubuntu:~/test/usbrelay-master# sudo make install_py
install -d /usr/lib/aarch64-linux-gnu
install -m 0755 libusbrelay.so /usr/lib/aarch64-linux-gnu
install -d /usr/bin
install -m 0755 usbrelay /usr/bin
python3 setup.py install
running install
running build
running build_ext
running install_lib
copying build/lib.linux-aarch64-3.5/usbrelay_py.cpython-35m-aarch64-linux-gnu.so -> /usr/local/lib/python3.5/dist-packages
running install_egg_info
Removing /usr/local/lib/python3.5/dist-packages/usbrelay_py-1.0.egg-info
Writing /usr/local/lib/python3.5/dist-packages/usbrelay_py-1.0.egg-info
root@tegra-ubuntu:~/test/usbrelay-master# python3 test.py 
Traceback (most recent call last):
  File "test.py", line 1, in <module>
    import usbrelay_py
ImportError: libusbrelay.so: undefined symbol: hid_error_

darrylb123, 2019-11-15:
My OS of choice is Fedora 31 so the Makefile needs to be modified to reflect the Python version on Fedora (3.7dm) and here is what happens when I do a build.
$ make clean; make; sudo make install; make python; sudo make install_py
rm -f usbrelay
rm -f libusbrelay.so
rm -f libusbrelay_py.so
rm -rf build
cc -shared -fPIC  -O2 -Wall -lhidapi-hidraw -L /usr/lib64 -Wl,-rpath /usr/lib64 -o libusbrelay.so libusbrelay.c
cc  -O2 -Wall usbrelay.c -lusbrelay -L./ -lhidapi-hidraw -L /usr/lib64 -Wl,-rpath /usr/lib64 -o usbrelay
install -d /usr/lib64
install -m 0755 libusbrelay.so /usr/lib64
install -d /usr/bin
install -m 0755 usbrelay /usr/bin
cc -shared -fPIC -I/usr/include/python3.7dm  -O2 -Wall -lhidapi-hidraw -L /usr/lib64 -Wl,-rpath /usr/lib64 -L./ -lusbrelay -o libusbrelay_py.so libusbrelay_py.c
python3 setup.py build
running build
running build_ext
building 'usbrelay_py' extension
creating build
creating build/temp.linux-x86_64-3.7
gcc -pthread -Wno-unused-result -Wsign-compare -DDYNAMIC_ANNOTATIONS_ENABLED=1 -DNDEBUG -O2 -g -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -fexceptions -fstack-protector-strong -grecord-gcc-switches -m64 -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -D_GNU_SOURCE -fPIC -fwrapv -fPIC -I/usr/include/python3.7m -c libusbrelay_py.c -o build/temp.linux-x86_64-3.7/libusbrelay_py.o
creating build/lib.linux-x86_64-3.7
gcc -pthread -shared -Wl,-z,relro -Wl,--as-needed -Wl,-z,now -g build/temp.linux-x86_64-3.7/libusbrelay_py.o -L./ -L/usr/lib -L/usr/lib64 -L/usr/lib/x86_64-linux-gnu -L/usr/lib/aarch64-linux-gnu -L/usr/lib/arm-linux-gnueabihf -L/usr/lib64 -lusbrelay -lpython3.7m -o build/lib.linux-x86_64-3.7/usbrelay_py.cpython-37m-x86_64-linux-gnu.so
install -d /usr/lib64
install -m 0755 libusbrelay.so /usr/lib64
install -d /usr/bin
install -m 0755 usbrelay /usr/bin
python3 setup.py install
running install
running build
running build_ext
running install_lib
copying build/lib.linux-x86_64-3.7/usbrelay_py.cpython-37m-x86_64-linux-gnu.so -> /usr/local/lib64/python3.7/site-packages
running install_egg_info
Removing /usr/local/lib64/python3.7/site-packages/usbrelay_py-1.0-py3.7.egg-info
Writing /usr/local/lib64/python3.7/site-packages/usbrelay_py-1.0-py3.7.egg-info
$ sudo python3 test.py
Count:  1
Boards:  (('PSUIS', 2, 0),)
Board:  ('PSUIS', 2, 0)
Result:  9
Result:  9
Result:  9
Result:  9

darrylb123, 2019-11-15:
Found the problem after installling an Ubuntu 16.04 VM and seeing the error.
The order of the object files in the Makefile is important on this version. It was not actually linking against libhidapi. 
I have tested it on 16.04 and Fedora and the change works on both.
Please clone the repository and try again.

darrylb123, 2019-11-15:
> Hello Darryl,
> I have tested with a new clone and it's working now.
> Thank you for your time and for your help!
>
> I will modify your test.py example and test the code with 2 boards to see
> if the strange behavior is still present.
> If you have another example of code except the test.py is really helpful
> (code that specify the exact board relay to be activated by giving the
> address "3D0V2").

Have a look at the earlier emails, there is a snippet of python code in one
that does just that.

closed 2019-12-01 by darrylb123