darrylb123/usbrelay #34: Ucreatefun.com - HIDRelay

beavis2k, 2020-01-11:
Bought a new relay with "ID 0519:2018 Star Micronics Co., Ltd"
but does not work, it does not detect the serial of the relay.
It uses a CH551G chip, red pcb with "QYF-UR0" marking.
[Aliexpress Listing](https://www.aliexpress.com/item/4000131078798.html)

Would it be possible to make this relay work with this code?
when running usbrelay:
`sudo USBID=0519:2018 usbrelay`
Device Found
  type: 0519 2018
  path: /dev/hidraw0
  serial_number: A0001
  Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 0

After searching online, I can control it with a raspberry pi and [hidapitester](https://github.com.cnpmjs.org/todbot/hidapitester)
`sudo ./hidapitester --list-detail`
0519/2018: Ucreatefun.com - HIDRelay
  vendorId:      0x0519
  productId:     0x2018
  usagePage:     0x0000
  usage:         0x0000
  serial_number: A0001
  interface:     0
  path: /dev/hidraw0

Hex Values
`0xF1 to F8` turns on relays 1-8
`0x01 to 08` turns off relays 1-8
`0xF9` turns on all relays
`0x09` turns off all relays

can be controlled by sending the hex value
`sudo ./hidapitester --vidpid 0519:2018 --open --send-output 0xF1 #on`
`sudo ./hidapitester --vidpid 0519:2018 --open --send-output 0x01 #off`

I only have a 1 channel version.
I hope this relay can be integrated to the project, if not, I hope it helps others with this relay.
Thanks for the library, the old relay works great with it.

darrylb123, 2020-01-12:
Looks interesting. I'll ser if I can get one.

beavis2k, 2020-01-12:
I am not sure how to read the state of the relay to fully control it, but since my needs are to turn on for less than a second and off simulating a PC power button press, its good enough.
Here is the [Seller?/Manufacturer?](https://m.1688.com/offer/583256081158.html) site
A [Japanese GitHub user](https://jj8xnp.github.io/AmateurRadio/USB_Relay_CW.html) that shows how to turn it on/off (bottom of page)
I purchased this relay because it was the cheapest one. Now I know to check if it has an AVR chip for it to be compatible with this program.

darrylb123, 2020-01-13:
Just ordered a 2 relay one. Do you think it has the equivalent of the id
serial. Can't see anything in your email.

darrylb123, 2020-03-05:
My relay arrived today. Looks like there is no way of differentiating multiple relays like the previous version. There is also no obvious way of getting the current status. Still, I'll incorporate the relay into my code.

beavis2k, 2020-03-06:
I used an old windows 7 Laptop I restored a few weeks ago and I finally decided to test the original software and code for the Ucreatefun relay. I found out the same thing, the program can only turn the relay on/off, cant read state or toggle the relay. Serial number is most likely the number of relays it has.

There is a video on [youtube](https://www.youtube.com/watch?v=a3XSIRkeubE) on how to reprogram the chip it runs on, hopefully some with more experience can dump and make a better firmware for this relay.

Also note that when sending the hex value, its case sensative, must be `0xF1` not `0xf1`

Thanks for adding the relay to the project.

darrylb123, 2020-03-08:
I've added the ability to operate the ucreatefun relays in the createfun_support branch.
Please:
git clone https://github.com/darrylb123/usbrelay.git
git checkout createfun_support
make
sudo make install

Test it
The ucreatefun relay serial is always A0001 but you can use the device path. Checkout the README for examples

beavis2k, 2020-03-08:
Git gave errors because the branch uses `-` not `_` in `createfun-support`
`laptop:~$ git clone https://github.com/darrylb123/usbrelay.git`
`Cloning into 'usbrelay'...`
`laptop:~$ cd usbrelay`
`laptop:~/usbrelay$ git checkout createfun-support`

lsusb shows `... ID 0519:2018 Star Micronics Co., Ltd`
I had a microsoft usb wireless mouse that was causing problems, it showed the mouse as hidraw 0, 1 and 2, leaving the relay as hidraw 3.  With hidapitester was able to see this
`laptop:~/hidapitester$ sudo ./hidapitester --list-detail`
> 045E/0745: Microsoft -
>   vendorId:      0x045E
>   productId:     0x0745
>   interface:     0
>   path: /dev/hidraw0

... repeats with interface 1 and 2 and /dev/hidraw1 and /dev/hidraw2

> 0519/2018: Ucreatefun.com - HIDRelay
>   vendorId:      0x0519
>   productId:     0x2018
>   serial_number: A0001
>   interface:     0
>   path: /dev/hidraw3

after removing the Microsoft mouse usb dongle, removing and re-insert the relay again works fine, it could be that the mouse is taking priority or confusing usbrelay.
With that mouse problem solved, it works great.
`laptop:~$ sudo usbrelay`

> Found 1 devices
> Device Found
>   type: 0519 2018
>   path: /dev/hidraw0
>   serial_number: A0001
>   Manufacturer: Ucreatefun.com
>   Product:      HIDRelay
>   Release:      1
>   Interface:    0
>   Number of Relays = 9
>   Module_type = 2

**relay 1 on**

`laptop:~$ sudo usbrelay A0001_1=1`
> Orig: A0001_1=1, Serial: A0001, Relay: 1 State: ff
> Found 1 devices
> Device Found
>   type: 0519 2018
>   path: /dev/hidraw0
>   serial_number: A0001
>   Manufacturer: Ucreatefun.com
>   Product:      HIDRelay
>   Release:      1
>   Interface:    0
>   Number of Relays = 9
>   Module_type = 2
> Serial: A0001, Relay: 1 State: ff
> 1 HID Serial: A0001 Serial: A0001, Relay: 1 State: ff
> target ff ucreate f1 f0 f0
> Serial: A0001, Relay: 1 State: ff --- Found

**relay 1 off**

`laptop:~$ sudo usbrelay A0001_1=0`

> Orig: A0001_1=0, Serial: A0001, Relay: 1 State: fd
> Found 1 devices
> Device Found
>   type: 0519 2018
>   path: /dev/hidraw0
>   serial_number: A0001
>   Manufacturer: Ucreatefun.com
>   Product:      HIDRelay
>   Release:      1
>   Interface:    0
>   Number of Relays = 9
>   Module_type = 2
> Serial: A0001, Relay: 1 State: fd
> 1 HID Serial: A0001 Serial: A0001, Relay: 1 State: fd
> target fd ucreate 1 f0 f0
> Serial: A0001, Relay: 1 State: fd --- Found

Its a single relay but tried to turn on/off with all relays
**all relays on/off**
`sudo usbrelay A0001_9=0` and `sudo usbrelay A0001_9=1` works fine.

with only the serial number we get
`laptop:~$ sudo usbrelay A0001`

> Orig: A0001, Serial: A0001, Relay: 0 State: 0
> Found 1 devices
> Device Found
>   type: 0519 2018
>   path: /dev/hidraw0
>   serial_number: A0001
>   Manufacturer: Ucreatefun.com
>   Product:      HIDRelay
>   Release:      1
>   Interface:    0
>   Number of Relays = 9
>   Module_type = 2
> Serial: A0001, Relay: 0 State: 0
> 1 HID Serial: A0001
> 
>  New serial can't be empty!
> 
> Serial: A0001, Relay: 0 State: 0 --- Not Found

`sudo usbrelay /dev/hidraw0_1=1` and `sudo usbrelay /dev/hidraw0_9=1` Also works fine

`sudo usbrelay A0001_1` assumes `sudo usbrelay A0001_1=0` turning off the relay if its on

Overall its working great.

darrylb123, 2020-03-09:
I've updated the doco and code to allow udev to create links to the relays by the physical port they are plugged into. I've only got 1 ucreatefun relay but it means that multiple ucreatefun relays can be used. I used 3 relays (1 ucreatefun and 2 dctech) to test and it works fine.

closed 2020-03-11 by darrylb123