darrylb123/usbrelay #35: renaming issue

vrm42, 2020-01-20:
Hi, I tried this amazing software because they sent me boards with similar serial numbers. I accidentally renamed one board to no name. Output is now:
_1=0
_2=0

usbrelay _0=NEWSN is not working. Is there any way to fix it?
Thanks!

darrylb123, 2020-01-21:
Have a look at an earlier bug. Has happened before.

vrm42, 2020-01-21:
Oh, sorry. I'm not familiar with all of these things, github, linux, compiling... didn't noticed other issues.
Now it works, took 2 hours so I backed up the recovery version of usbrelay :) Thanks!

closed 2020-01-21 by vrm42

darrylb123, 2020-01-21:
Would you mind looking at your shell history and post the command that
caused the problem. Type 'history' to get it.

vrm42, 2020-01-21:
This is it, but I don't think it's correct. Some lines make no sense, like 19 and 20. I typed them later for sure, but I got two boards with HW341, SPN01 was successfull, tried SPN02 which caused the issue. I've renamed ~20 boards in last couple of months even those that got non ascii names. Orange pi is newly deployed. Usbrelay I downloaded with wget is not the most recent version but it worked before for renaming.

1  armbian-config
    2  reboot
    3  wget
    4  mkdir
    5  mkdir /st
    6  cd /st
    7  wget http://192.168.1.93/ewbf/usbrelay
    8  ls
    9  chmod 777 usbrelay
   10  ls -l
   11  ./usbrelay
   12  wget http://192.168.1.93/ewbf/trash/usblib.deb
   13  apt-get install usblib.deb
   14  dpkg -i usblib.deb
   15  top
   16  htop
   17  reboot
   18  usbrelay
   19  usbrelay _0=SPN01
   20  usbrelay _0=SPN01
   21  usbrelay
   22  usbrelay _1=1
   23  usbrelay | grep serial_number
   24  usbrelay
   25  usbrelay HW341_0=SPN01
   26  usbrelay
   27  usbrelay
   28  usbrelay ""_0=SPN02
   29  usbrelay
   30  usbrelay _1=1
   31  usbrelay _1=1
   32  usbrelay
   33  usbrelay _1=1
   34  vi sn
   35  usbrelay $(cat sn)_0=SPN02
   36  usbrelay
   37  vi sn
   38  usbrelay $(cat sn)_0=SPN02

tried nothing and then alt+255 :)

darrylb123, 2020-01-22:
Did you do it as root? You obviously didn't use sudo...
That would explain it.

darrylb123, 2020-01-22:
You should use the github source and compile it.
The dock on github has instructions for compiling it. It's very simple.