darrylb123/usbrelay #39: Wildcard Disabling Ability to Use Relay

jacksoncejda, 2020-03-05:
Originally in my setup I had one 8-channel usbrelay device and there were no issues. However, I recently added a second 8-channel usbrelay to my setup, and now there are wildcards in my device ID's. Here is the result of running $ usbrelay:
```
$ usbrelay
Device Found
  type: 16c0 05df
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
?#}??_1=0
?#}??_2=0
?#}??_3=0
?#}??_4=0
?#}??_5=0
?#}??_6=0
?#}??_7=0
?#}??_8=0

Device Found
  type: 16c0 05df
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
?QMBS_1=0
?QMBS_2=0
?QMBS_3=0
?QMBS_4=0
?QMBS_5=0
?QMBS_6=0
?QMBS_7=0
?QMBS_8=0
```

The wildcards seem to be prohibiting my ability to turn on and off the relays. When I run:
```
$ usbrelay ?QMBS_1=1
```
the output is:
```
$ usbrelay ?QMBS_1=1
Orig: ?QMBS, Serial: ?QMBS, Relay: 1 State: ff
Device Found
  type: 16c0 05df
  path: /dev/hidraw1
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
Serial: ?QMBS, Relay: 1 State: ff 

Device Found
  type: 16c0 05df
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
Serial: ?QMBS, Relay: 1 State: ff 

Serial: ?QMBS, Relay: 1 State: ff --- Not Found
```

I have tried following your doco to change the serial, but it am not having luck because typing the wildcard is not the same as the correct serial number.

Any suggestions on how to move forward?

darrylb123, 2020-03-06:
Please use the latest version. I think you're problem is that you need to
use sudo or run as root.

Darryl

jacksoncejda, 2020-03-06:
The problem persists after using sudo. I have also tried it on Ubuntu computer as a root user and got the same results.

darrylb123, 2020-03-06:
Does it work properly with either of the relays plugged in by themselves?
If so, note the serials and use those when both plugged in to see if they
still work.
Have you built the latest code or were you using apt-get which is quite old.

Darryl

jacksoncejda, 2020-03-06:
When the relays are plugged in by themselves I still get the same result. However, this problem only started after plugging in the second relay. Before that, the first relay worked without issue.

Originally I was using apt-get, but while debugging this issue, I have built with the most recent code. No change in result.

darrylb123, 2020-03-07:
Can you escape the wildcard characters and try to operate them. Is the
shell removing the ? chars.

In the past I have advised someone to change the code so that any device
can be renamed.
Have a look at Issue #26. It shows which lines to comment out and
recompile. Then with one device at a time, try to change their serial.

I really have no clue how the second one plugged in changed the first. Can
you look at your shell history and send all your usbrelay commands?
$ history | grep usbrelay

Darryl

mithunnj, 2020-03-09:
The solution to this is to rename the serial by calling it by the /dev/hidraw* route instead of the serial name:

In our case, the serial was called ?QMBS, and we wanted to rename it ABCDE - we tried the following:
`sudo ./usbrelay ?QMBS_0=ABCDE` would fail. But the usb was plugged to `/dev/hidraw0` and the following worked: `sudo ./usbrelay /dev/hidraw0_0=ABCDE`

closed 2020-03-09 by jacksoncejda

darrylb123, 2020-03-10:
Only posted that feature last night.  Very quick to find it. :)

Darryl

mithunnj, 2020-03-10:
Cheers mate!

Thanks, 
Mit