darrylb123/usbrelay #42: *** buffer overflow detected ***: ./usbrelay terminated Aborted (core dumped)

mithunnj, 2020-03-10:
I suspect that the FORTIFY SOURCE gcc compiler flag is raising a warning after adding a check code into certain calls that could cause buffer overflows.

As described [here](https://github.com/intel/safestringlib/wiki/FORTIFY-SOURCE-and-Safe-String-Library) FORTIFY SOURCE works on only a few functions, including: memcpy, mempcpy, memmove, memset, strcpy, stpcpy, strncpy, strcat, strncat, sprintf, vsprintf, snprintf, vsnprintf, gets.

mithunnj, 2020-03-10:
Additional useful information about FORTIFY SOURCE macro: https://access.redhat.com/blogs/766093/posts/1976213

mithunnj, 2020-03-10:
[image]

Result of a core dump after the *buffer overflow* error.

mithunnj, 2020-03-10:
As described in the following: 
```
#9  0x00007ffff7911fcb in ___vsprintf_chk (s=0x7fffffffdd60 "HP USB 1000dpi LaseUUU", flags=1, slen=20, format=0x7ffff7bd2aba "%ls", 
    args=args@entry=0x7fffffffdc80) at vsprintf_chk.c:82
#10 0x00007ffff7911efa in ___sprintf_chk (s=<optimized out>, flags=<optimized out>, slen=<optimized out>, format=<optimized out>)
    at sprintf_chk.c:31
```
I suspect that the the character array `HP USB 1000dpi LaseUUU` results in an overflow because it exceeds `slen=20`

darrylb123, 2020-03-10:
Thanks, it seems USB product names can be up to 255 bytes. (Actually 128 bytes after UTF16)
I don't have any USB devices with long product names can you test the new version?

closed 2020-03-18 by darrylb123