darrylb123/usbrelay #47: 2 Ucreatefun.com on same Rpi => no use of second Relay board

sebbonnaire, 2020-04-17:
Hi Team !

First of wall, let me congratulate you for your work on the "usbrelay" project. I use it daily, and thanks to its stability, i'm confident with the automation at home scale. So, i bought a few more relay cards (2 cards with 8 relay each)

And i get 2 "Ucreatefun.com" 8 relays cards... connected now on an Rpi B version 1.
lsusb says : 
```
Bus 001 Device 006: ID 0519:2018 Star Micronics Co., Ltd
Bus 001 Device 005: ID 0519:2018 Star Micronics Co., Ltd

```

My only problem is that when i use this command :
`sudo usbrelay /dev/hidraw0_1=1`
or this command : 
`sudo usbrelay /dev/hidraw1_1=1`
i've got the exact same result : the "first" board is changing is relay1 state... even if i try to change the second "hidraw" board...

may you help me to use this command as i should or point me to changes i may try on the source code.

for information, i'm using current git code compiled today (17-04-2020).

Regards

darrylb123, 2020-04-17:
I only have one of the ucreatefun devices

Can you attach the output of
$ sudo usbrelay -d

Darryl

sebbonnaire, 2020-04-18:
> I only have one of the ucreatefun devices Can you attach the output of $ sudo usbrelay -d Darryl
> […](#)

Hi Darryl,
Thank you for your quick reply.
As requested, here is the command output. 
Best regards
```
sudo usbrelay -d
Found 2 devices
Device Found
  type: 0519 2018
  path: /dev/hidraw0
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1
Device Found
  type: 0519 2018
  path: /dev/hidraw1
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1
```

darrylb123, 2020-04-18:
Do they work correctly when each one is plugged in by itself?

sebbonnaire, 2020-04-18:
Oh yes ! 
The two cards are working if each one is the only one connected

darrylb123, 2020-04-18:
> When both modules are plugged in, does
> usbrelay /dev/hidraw1_1=1 operate both modules or always the first module
> only?
>
> Can you run
> usbrelay -d /dev/hidraw0_1=1
> usbrelay -d /dev/hidraw1_1=1
>
> And attach the output.

Sorry for all the questions, but what os are you running on your pi?
I have a pi running raspian. My testing as always been on x86_64.

sebbonnaire, 2020-04-18:
Hi

I'm really happy to help, and i can fully understand the amount of question.
So, first of all, yes i'm using a Pi with Raspian
```
uname -a
Linux raspberrypi 4.19.97+ #1294 Thu Jan 30 13:10:54 GMT 2020 armv6l GNU/Linux

```

Then, when i run this command `usbrelay /dev/hidraw1_1=1`, it does operate always the first module (first in the order or USB ports apparently). Because when i witch the USB cable from their ports, then the "other card" lights on.

here is the command output :
```
usbrelay -d /dev/hidraw0_1=1
Orig: /dev/hidraw0_1=1, Serial: /dev/hidraw0, Relay: 1 State: ff
Found 2 devices
Device Found
  type: 0519 2018
  path: /dev/hidraw0
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1
Device Found
  type: 0519 2018
  path: /dev/hidraw1
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1
Serial: /dev/hidraw0, Relay: 1 State: ff
Serial: /dev/hidraw0, Relay: 1 State: ff
```

and the second one : 

```
usbrelay -d /dev/hidraw1_1=1
Orig: /dev/hidraw1_1=1, Serial: /dev/hidraw1, Relay: 1 State: ff
Found 2 devices
Device Found
  type: 0519 2018
  path: /dev/hidraw0
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1
Device Found
  type: 0519 2018
  path: /dev/hidraw1
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1
Serial: /dev/hidraw1, Relay: 1 State: ff
Serial: /dev/hidraw1, Relay: 1 State: ff
```

darrylb123, 2020-04-18:
I can't see anything out of place there.
Can you compile and test on an x86 computer. Just want to discount the RPI.

darrylb123, 2020-04-19:
I cannot duplicate your error with a mix of the 2 types of relays. I had a thought that I have used libusb instead of hidraw to fix issues in the past. 
I have modified the code so that the libusb path can be used so you will need to pull my changes. I have updated the README to show how to use it.
$ sudo usbrelay -d
Device Found
  type: 0519 2018
  path: 0001:0015:00
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2

$ sudo ./usbrelay 0001:0015:00_1=0

See if that helps.
I need to play with libusb some more as the device changes after each unplug/plug. There must be a way to fix the paths.

sebbonnaire, 2020-04-19:
Hi Darryl,

here is my first result. I'm still using the Rpi as i do not have (yet) a regular PC with linux installed.
```
./usbrelay -d
Found 1 devices
Device Found
  type: 0519 2018
  path: 0001:0006:00
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1
Device Found
  type: 0519 2018
  path: 0001:0005:00
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1
corrupted size vs. prev_size
Aborted
root@raspberrypi:~/usbrelay# ./usbrelay 0001:0006:00_2=1
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1
free(): invalid next size (fast)
Aborted
```

darrylb123, 2020-04-19:
That looks nasty. I probably won't get back to it for a couple of days.
I'll have a look in the mean time.

darrylb123, 2020-04-19:
Just a quick test
In libusbrelay.h change
#define Serial_Length 5
To
#define Serial_Length 20

And recompile

sebbonnaire, 2020-04-20:
Hi Darryl,

I've installed a linux on an old PC i have here (an 32 Bits Athlon XP 3000+, almost 20 years old already). By the way, i could install an Ubuntu server on it, and install usbrelay package.

I can now confirm that the "addressing" problem is the same on a x86 computer as on a RaspberryPi : only one card (depending on the USB port) can be controlled with the following commands (the number of "hidraw" doesnt matter) : 
usbrelay /dev/hidraw0_1=1
usbrelay /dev/hidraw1_1=1

If having access to my Rpi can help, tell me via e-mail.

darrylb123, 2020-04-20:
I would appreciate being able to log in to your RPI if necessary. I'm in
eastern Australia so our timezones might not be compatible...
In the mean time, have a look at closed issue 34 where ii was introduced to
the createfun relay. The submitter could operate the relay with
hidapitester. His examples used the vpid which would be no good with 2
modules but I notice that there is the option to use --open-path
/dev/hidrawx. I built hidapitester and tried his examples when my relay
arrived.

I don't understand why hidraw is not accepting the path to the device. It
might be a bug in my code (most likely) or some strangeness in hidapi.

If you can try hidapitester with 2 relays attached and see if you can
operate the relays correctly.

I'd really appreciate the testing.

Darryl

darrylb123, 2020-04-21:
I have added some extra debugging. Please do a git pull and make/sudo make install as hidraw and test with -d and post the results.
ie 
git pull
make 
sudo make install
sudo usbrelay -d /dev/hidraw0_1=1 /dev/hidraw1_1=1

sebbonnaire, 2020-04-21:
Hi Darryl

For the access to my Rpi, send me a mail directly.
I can imagine wake up early, it's not an issue to me.

I just have tried the "extra debugging" version.
As you'll see, the "Operate path" is always the same in the output log
`Operate /dev/hidraw0 path`

here is the full output : 
```
usbrelay -d /dev/hidraw0_1=1 /dev/hidraw1_1=1
Orig: /dev/hidraw0_1=1, Serial: /dev/hidraw0, Relay: 1 State: ff
Orig: /dev/hidraw1_1=1, Serial: /dev/hidraw1, Relay: 1 State: ff
Found 2 devices
Device Found
  type: 0519 2018
  path: /dev/hidraw0
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1
Device Found
  type: 0519 2018
  path: /dev/hidraw1
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1
Serial: /dev/hidraw0, Relay: 1 State: ff
Serial: /dev/hidraw0, Relay: 1 State: ff
Operate /dev/hidraw0 path
Serial: /dev/hidraw1, Relay: 1 State: ff
Serial: /dev/hidraw1, Relay: 1 State: ff
Operate /dev/hidraw0 path
Serial: /dev/hidraw1, Relay: 1 State: ff --- Found
```

darrylb123, 2020-04-21:
Fantastic, so it's in my program. I'll see what can find.

darrylb123, 2020-04-22:
Think I have it.
You'll see a lot more debugging. I was passing the serial to the operate, rather than the device. It would therefore operate the first serial A0001 that matched. Now uses the device all the way though.
Can you:
git pull
make
sudo make install
sudo usbrelay -d /dev/hidraw0_1=1 /dev/hidraw1_1=1

sebbonnaire, 2020-04-22:
Hi Darryl,

here is the output of the command given (still one only board is acting) : 
```
usbrelay -d /dev/hidraw0_1=0 /dev/hidraw1_1=0
Orig: /dev/hidraw0_1=0, Serial: /dev/hidraw0, Relay: 1 State: fd
Orig: /dev/hidraw1_1=0, Serial: /dev/hidraw1, Relay: 1 State: fd
enumerate_relay_boards()Found 2 devices
Device Found
  type: 0519 2018
  path: /dev/hidraw0
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1
Device Found
  type: 0519 2018
  path: /dev/hidraw1
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1
Serial: /dev/hidraw0, Relay: 1 State: fd
Serial: /dev/hidraw0, Relay: 1 State: fd
find_board(A0001) path /dev/hidraw0
operate_relay(A0001,) /dev/hidraw0 path
Serial: /dev/hidraw1, Relay: 1 State: fd
Serial: /dev/hidraw1, Relay: 1 State: fd
find_board(A0001) path /dev/hidraw0
operate_relay(A0001,) /dev/hidraw0 path
Serial: /dev/hidraw1, Relay: 1 State: fd --- Found
```

darrylb123, 2020-04-22:
You don't seem to have all the debugging so I assume it's not fully built.
Do a:
make clean
git pull
make
sudo make install

And try again

sebbonnaire, 2020-04-22:
you are right, i had to "make clean". my bad. Sorry !

here is the output now : it works 🥇 
```
./usbrelay -d /dev/hidraw0_1=1 /dev/hidraw1_1=1
Orig: /dev/hidraw0_1=1, Serial: /dev/hidraw0, Relay: 1 State: ff
Orig: /dev/hidraw1_1=1, Serial: /dev/hidraw1, Relay: 1 State: ff
enumerate_relay_boards()Found 2 devices
Device Found
  type: 0519 2018
  path: /dev/hidraw0
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1
Device Found
  type: 0519 2018
  path: /dev/hidraw1
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1
main() arg 0 Serial: /dev/hidraw0, Relay: 1 State: ff
find_board(/dev/hidraw0) path /dev/hidraw0
main() operate: /dev/hidraw0, Relay: 1 State: ff
find_board(/dev/hidraw0) path /dev/hidraw0
operate_relay(/dev/hidraw0,) /dev/hidraw0 path
main() arg 1 Serial: /dev/hidraw1, Relay: 1 State: ff
find_board(/dev/hidraw1) path /dev/hidraw1
main() operate: /dev/hidraw1, Relay: 1 State: ff
find_board(/dev/hidraw1) path /dev/hidraw1
operate_relay(/dev/hidraw1,) /dev/hidraw1 path
Serial: /dev/hidraw1, Relay: 1 State: ff --- Found
```

darrylb123, 2020-04-22:
Great.
Have a look at the docs how you can create persistent links using udev. You
can set up physical port references in lieu of their lack of unique
serial.l numbers.
Thanks for finding the bug. I shall close the issue.

Darryl

darrylb123, 2020-04-22:
If you wouldn't mind, can we just confirm that libusb works now with those relays. I can't fault it here
make clean
make HIDAPI=libusb
sudo make install

You will need to substitute your paths: mine looks like this:
sudo usbrelay -d 0003:0006:00_1=1 0003:000a:00_2=0 0003:0005:00_1=1 0003:0008:00_2=1

sebbonnaire, 2020-04-24:
hi Darryl

sorry for my late answer, but i was "quite" busy with my work those days.
here is the output of the command `usbrelay-d` with libusb : 
```
./usbrelay -d
enumerate_relay_boards()Found 1 devices
Device Found
  type: 0519 2018
  path: 0001:0006:00
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1
double free or corruption (out)
Aborted
```

darrylb123, 2020-04-24:
OK, I will try myself with an 8 relay dcctech module. I can't duplicate
with 2 relay modules of either type.

Thanks for your trouble.
Can you raise a new issue about libusb errors. I don't want to mix with
your initial problem.

Darryl

sebbonnaire, 2020-04-24:
ok

a big thank your for your help on this case Darryl !

closed 2020-04-24 by sebbonnaire