darrylb123/usbrelay #49: usbrelay show no relay serial

felipeAH, 2020-04-22:
Hi,

session transcript follows, 

felipe@satriani:~$ sudo usbrelay
Device Found
  type: 16c0 05df
  path: /dev/hidraw5
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
_1=0
_2=0
_3=0
_4=0
_5=0
_6=0
_7=0
_8=0

as you can see, no relay serial is displayed .. so no relay can be managed. I've tried this:

felipe@satriani:~$ sudo usbrelay '_1'=1
Orig: _1, Serial: 1, Relay: 1 State: 0
Device Found
  type: 16c0 05df
  path: /dev/hidraw5
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
Serial: 1, Relay: 1 State: 0 

Serial: 1, Relay: 1 State: 0 --- Not Found
felipe@satriani:~$ sudo usbrelay
Device Found
  type: 16c0 05df
  path: /dev/hidraw5
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
_1=0
_2=0
_3=0
_4=0
_5=0
_6=0
_7=0
_8=0

tested with versions: v0.4, 0.5, 0,6 same result

version 0.7 show nothing

felipe@satriani:~$ sudo usbrelay
felipe@satriani:~$

version v0.4 work with other relay, same chipset only 1 Relay on the board.

thank you.

darrylb123, 2020-04-23:
Please get the very latest version. There has been a lot of work in using
devices to control the relays. You can set a new serial using the
/dev/hidrawx device.

darrylb123, 2020-04-23:
Just to make sure there is no confusion. Please clone and build the latest code from github.
The README shows you how to do it and how to use the devices.
https://github.com/darrylb123/usbrelay.git

darrylb123, 2020-04-23:
> HI again,
>
> I've build the latest and can set the serial, set value and access the
> board via /dev/hidraw device, but relays does not switch, ( no click sound
> ).
>
> session transcript
>
> felipe@satriani:$ sudo LD_LIBRARY_PATH=. ./usbrelay /dev/hidraw5_1=1
> _1=0
> _2=0
> _3=0
> _4=0
> _5=0
> _6=0
> _7=0
> _8=0
>
> felipe@satriani:$ sudo LD_LIBRARY_PATH=. ./usbrelay /dev/hidraw5_0=ZAQ12
> _1=1
> _2=0
> _3=0
> _4=0
> _5=0
> _6=0
> _7=0
> _8=0
> Setting new serial
>
> felipe@satriani:$ sudo LD_LIBRARY_PATH=. ./usbrelay
> ZAQ12_1=1
> ZAQ12_2=0
> ZAQ12_3=0
> ZAQ12_4=0
> ZAQ12_5=0
> ZAQ12_6=0
> ZAQ12_7=0
> ZAQ12_8=0
>
> tried all 8 relays .. same result. No click.
>
> thank you.

I have an 8 relay device which works.. The relays do not operate without
the additional 12vdc supply. Has this module worked before?

Can you try the windows program and see if that works?

felipeAH, 2020-04-23:
> I have an 8 relay device which works.. The relays do not operate without the additional 12vdc supply.

maybe that's the thing .. i'll try with a power supply and tell you.

 Has this module worked before? Can you try the windows program and see if that works?
> […](#)

Can you send me the Windows Program link to dowload it?  the board came alone.

Thank you very much.

darrylb123, 2020-04-23:
Its usually linked to on the eBay item. Did you buy it there?

felipeAH, 2020-04-23:
Hi,

i buyed it at aliexpress

darrylb123, 2020-04-24:
Aliexpress say they will provide the software. Ask them for it. I don't
have it.
Did you try the module with 12v applied?

darrylb123, 2020-04-24:
Filipe,
I have a 8 relay DCCTECH module. I just tested it without 12V and
everything works except the relay itself does not energise.
$ sudo usbrelay -d HW554_5=1
Orig: HW554_5=1, Serial: HW554, Relay: 5 State: ff
enumerate_relay_boards()Found 1 devices
Device Found
  type: 16c0 05df
  path: /dev/hidraw4
  serial_number: HW554
Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
  Module_type = 1
HW554_1=0
HW554_2=0
HW554_3=0
HW554_4=0
HW554_5=0
HW554_6=0
HW554_7=0
HW554_8=0
main() arg 0 Serial: HW554, Relay: 5 State: ff
find_board(HW554) path /dev/hidraw4
main() operate: HW554, Relay: 5 State: ff
find_board(HW554) path /dev/hidraw4
operate_relay(HW554,) /dev/hidraw4 path
$ sudo usbrelay -d
enumerate_relay_boards()Found 1 devices
Device Found
  type: 16c0 05df
  path: /dev/hidraw4
  serial_number: HW554
Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
  Module_type = 1
HW554_1=0
HW554_2=0
HW554_3=0
HW554_4=0
HW554_5=1                    <---- Note that the 5th relay indicates on
HW554_6=0
HW554_7=0
HW554_8=0

Darryl

felipeAH, 2020-04-25:
Hi darryl,

same results here,  should be the lack of power, unfortunatelly i dont't have 
a power supply for it at the moment. 

thanks for your help.

best regards.

closed 2020-04-26 by darrylb123