darrylb123/usbrelay #50: libusb is not usable for 2 Ucreatefun relay boards

sebbonnaire, 2020-04-24:
hi Darryl

as requested, here is the case with the libusb...
here is the output of the command `usbrelay-d` with libusb, on a Rpi, with 2 x 8 relay boards attached : 
```
./usbrelay -d
enumerate_relay_boards()Found 1 devices
Device Found
  type: 0519 2018
  path: 0001:0006:00
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1
double free or corruption (out)
Aborted
```

darrylb123, 2020-04-24:
I tested with an 8 relay DCCTECH module. No issues like you have seen.
I may need to log into your rPi unless you are familiar with gdb. Send an
email to darryl . bond at gmail . com
and we will organise some keys.
sudo usbrelay -d
enumerate_relay_boards()Found 1 devices
Device Found
  type: 16c0 05df
  path: 0003:0009:00
  serial_number: HW554
Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
  Module_type = 1
HW554_1=0
HW554_2=0
HW554_3=1
HW554_4=0
HW554_5=0
HW554_6=0
HW554_7=0
HW554_8=0

darrylb123, 2020-04-27:
Diagnosis showed that there was a /usr/sbin/usbrelay that was an old version and the subsequent calls to the new libusbrelay.so caused wierdness.
Deleted /usr/sbin/usbrelay and pulled the very latest version and rebuilt and reinstalled. Now
# usbrelay -d
enumerate_relay_boards()Found 2 devices
Device Found
  type: 0519 2018
  path: 0001:0006:00
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1
Device Found
  type: 0519 2018
  path: 0001:0005:00
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1

darrylb123, 2020-04-27:
Added git version numbers to the executable and library that is printed out in debug mode.
The versions can be easily compared.# usbrelay -d
Version: 0175305e57f3ae0a193df2edb894f343dda90702
Library Version: 0175305e57f3ae0a193df2edb894f343dda90702
enumerate_relay_boards()Found 2 devices
Device Found
  type: 0519 2018
  path: 0001:0006:00
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1
Device Found
  type: 0519 2018
  path: 0001:0005:00
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1

darrylb123, 2020-04-27:
Versions of the library and executable now printed in debug mode
# usbrelay -d
Version: 0175305e57f3ae0a193df2edb894f343dda90702
Library Version: 0175305e57f3ae0a193df2edb894f343dda90702
enumerate_relay_boards()Found 2 devices
Device Found
  type: 0519 2018
  path: 0001:0006:00
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1
Device Found
  type: 0519 2018
  path: 0001:0005:00
  serial_number: A0001
Manufacturer: Ucreatefun.com
  Product:      HIDRelay
  Release:      1
  Interface:    0
  Number of Relays = 9
  Module_type = 2
A0001_1=-1
A0001_2=-1
A0001_3=-1
A0001_4=-1
A0001_5=-1
A0001_6=-1
A0001_7=-1
A0001_8=-1
A0001_9=-1

closed 2020-04-27 by darrylb123

sebbonnaire, 2020-04-27:
Thanks you Darryl, this version information was missing. You are right