darrylb123/usbrelay #52: AUR package of USBrelay fails to build on Manjaro 5.4

ESPWarren, 2020-05-15:
It has multiple failures. First it was the MDS for master.zip - I edited the build script.

That sorted the mds but then it failed further on - log:

  GNU nano 4.9.2                          g                                     
Cloning usbrelay build files...
Checking usbrelay dependencies...
Resolving dependencies...
Checking inter-conflicts...

Building usbrelay...
==> Making package: usbrelay 20190629-1 (Fri 15 May 2020 17:05:01 BST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Found master.zip
==> Validating source files with md5sums...
    master.zip ... Passed
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Extracting master.zip with bsdtar
==> Starting prepare()...
==> Starting build()...
make: *** No rule to make target '.git/HEAD', needed by 'gitversion.c'.  Stop.
==> ERROR: A failure occurred in build().

Here is the build script:

# Maintainer: Your Name <linuxboy@fel.hopto.org>
pkgname=usbrelay
pkgver=20190629
pkgrel=1
pkgdesc="Control usb relay - based on hidapi"
url="https://github.com/darrylb123/usbrelay"
license=('GPL-2.0')
groups=('system')
depends=('hidapi')
arch=('i686' 'x86_64')
makedepends=()
checkdepends=()
optdepends=()
provides=()
conflicts=()
replaces=()
backup=()
options=()
install=
changelog=
source=("https://github.com/darrylb123/usbrelay/archive/master.zip")
noextract=()
md5sums=('B42B221CC51B5FEA0D22661E379A92FE')
validpgpkeys=()

prepare() {
        cd "$pkgname-master"
}

build() {
        cd "$pkgname-master"
        make
}

package() {
        cd "$pkgname-master"
        make DESTDIR="$pkgdir/" install
}

darrylb123, 2020-05-15:
Oh dear,
The zip file does not include the git metadata to be able to determine the version.
I have no clue about Manjaro,
Can you use git clone instead of the zip file? That would solve your problem.

I will think about how to make the zip file compile.

darrylb123, 2020-05-16:
I have changed the way the version file is created. It is now part of the zip file and so should build on Manjaro

ESPWarren, 2020-05-16:
Thanks for the responsiveness - this really helps...

Ok that is obviously a move in the right direction - it no longer throws an MD5 error with the main zip file.

But it still has an issue:

Preparing...
Cloning usbrelay build files...
Checking usbrelay dependencies...
Resolving dependencies...
Checking inter-conflicts...

Building usbrelay...
==> Making package: usbrelay 20190629-1 (Sat 16 May 2020 14:06:08 BST)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Found master.zip
==> Validating source files with md5sums...
    master.zip ... Passed
==> Removing existing $srcdir/ directory...
==> Extracting sources...
  -> Extracting master.zip with bsdtar
==> Starting prepare()...
==> Removing existing $pkgdir/ directory...
==> Starting build()...
make: *** No rule to make target '.git/HEAD', needed by 'gitversion.c'.  Stop.
==> ERROR: A failure occurred in build().
    Aborting...

darrylb123, 2020-05-16:
There is some old code somewhere. There is no reference to git in the
makefile anymore.

wentasah, 2020-05-17:
Hi, I don't think it is a good idea for a distribution to build the package directly from master. First, as @ESPWarren  points out, there will be problems with checksums whenever a new commit is pushed. Second, some versions might not be stable. For example, few commits back, one of my relays stopped working. Fortunately, Darryl fixed it before I found time to file an issue :-).

So I think we should do the following (I'll send PR if @darrylb123 is not against):
- If `gitversion.c` is missing during build, Makefile should generate one with version `???` or something like this.
- We should provide release zip/tar.gz files with generated `gitversion.c`, which corresponds to the commit. These will be distributed as release assets. Darryl already did that for 0.7. I think I'll extend the Makefile to generate these release zip/tar, for example by running `make release`. And uploading the assets can be automated. So maybe, we can also have `make upload-assets`. Alternatively, this all could by automated with [GitHub Action](https://github.com/marketplace/actions/upload-a-release-asset).

severach, 2020-06-08:
AUR package is fixed.

ESPWarren, 2020-06-15:
Thanks Chris

Worked like a charm!

Nick

closed 2020-06-26 by ESPWarren