darrylb123/usbrelay #56: Glitched serials?

ESPWarren, 2020-06-26:
Hi

I have an 8 relay board.

When I use usbrelay, it reports 8 relay states but with glitched serials:
ÿÿÿÿÿ_1=0
ÿÿÿÿÿ_2=0
ÿÿÿÿÿ_3=0
ÿÿÿÿÿ_4=0
ÿÿÿÿÿ_5=0
ÿÿÿÿÿ_6=0
ÿÿÿÿÿ_7=0
ÿÿÿÿÿ_8=0

I am running it on a raspberry pi

How do I access the relays in this situation?

darrylb123, 2020-06-26:
When I've seen that before, it's because sudo wasn't used. You need to have
write permission to the device.

ESPWarren, 2020-06-27:
I was using sudo!

Here is the full output when I use "sudo usbrelay":

Device Found
  type: 16c0 05df
  path: /dev/hidraw0
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8

ÿÿÿÿÿ_1=0
ÿÿÿÿÿ_2=0
ÿÿÿÿÿ_3=0
ÿÿÿÿÿ_4=0
ÿÿÿÿÿ_5=0
ÿÿÿÿÿ_6=0
ÿÿÿÿÿ_7=0
ÿÿÿÿÿ_8=0

This is what the board looks like:

[image]

darrylb123, 2020-06-27:
If you build the latest version, you can change the serial using the
device. Note the device name shown when you run usbrelay.
Eg in your case:
sudo usbrelay /dev/hidraw0_0=AAAAA

ESPWarren, 2020-06-27:
OK I had installed this on my Pi using:
`
sudo apt-get install usbrelay`

It reports it being version 0.4-1

Is there something else I have to build to be using 'the latest version'?

I am assuming that apt-get install would be using the latest version...

When I try your suggestion:

`sudo usbrelay /dev/hidraw0_0=AAAAA`

I get:

***buffer overflow detected***; usbrelay aborted

??

darrylb123, 2020-06-27:
The version installed by apt-get is quite old.
Please follow the instructions for building usbrelay on the GitHub page.
You found this to register this issue.

ESPWarren, 2020-06-28:
OK,

I uninstalledd the apt version (sudo apt-get remove usbrelay)
I followed your instructions:

$ sudo apt-get install libhidapi-dev libhidapi-hidraw0 git 
$ git clone https://github.com/darrylb123/usbrelay
$ cd usbrelay
$ make
$ sudo make install

The installation went ahead perfectly - it told me that lidhidapi-hidraw0 is already the newest version.

usbrelay -V tells me the version I am now using:

usbrelay 0.7-33-gf3434b9050

Sadly, sudo usbrelay  gave me the precise same glitched serials on the relays as previously....

Possibilities:

1) The apt-get uninstall did not properly remove my original version
2) Maybe my hidraw library is not actually the latest
3) I just need to start wth a completely clean install on the PI with a fresh raspbian image
4) My relay board is non-standard...

Thoughts?

darrylb123, 2020-06-28:
Assuming the serial is glitched, change it to something you want. As I
described, you can change it using the device with the latest version. Look
in the doco for a full description.

glunardi, 2020-07-19:
I believe I have the board from the same manufacturer. Purchased here a few days ago:
https://www.amazon.com/gp/product/B07DYNVB38

The board works great out of the box with Ubuntu 20.04 (machine came with it):
```
Package: usbrelay
Version: 0.4-2
Filename: pool/universe/u/usbrelay/usbrelay_0.4-2_amd64.deb
```

With the right udev rule installed, regular user or root outputs:
```
Device Found
  type: 16c0 05df
  path: /dev/hidraw14
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
6QMBS_1=0
6QMBS_2=0
6QMBS_3=0
6QMBS_4=0
6QMBS_5=0
6QMBS_6=0
6QMBS_7=0
6QMBS_8=0
```

Installed from source here (to gain access to the python library). With `0.7-33-gf3434b9050` I get this output:
```
Version: 0.7-33-gf3434b9050
Library Version: 0.7-33-gf3434b9050
enumerate_relay_boards()Found 1 devices
Device Found
  type: 16c0 05df
  path: /dev/hidraw14
  serial_number: 6QMBS
Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
  Module_type = 1
6QMBS_1=0
6QMBS_2=0
6QMBS_3=0
6QMBS_4=0
6QMBS_5=0
6QMBS_6=0
6QMBS_7=0
6QMBS_8=0
```

The relays work fine (only powered over USB) and the leds light up. In other words, I am really happy with that little board and even more with your project. Thank you very much @darrylb123 for saving me the time of figuring out the usb raw hid for this board. Awesome work!

closed 2020-09-14 by darrylb123