darrylb123/usbrelay #57: Activate All Relays

thunkableBob, 2020-08-25:
What is the entry to enable all relays at once?
I have tried relay_9=1  and relay_0=1.  (0 changes serial to 1).
I changed serial from 6QMBS to relay.
I can successfully activate each relay, but need to activate all at once.

darrylb123, 2020-08-25:
Should be possible, let me look into it.

darrylb123, 2020-08-26:
What about putting them all on the one command line, does that operated
close enough together?

darrylb123, 2020-09-01:
Added the mythical relay 9 to operate all relays at once. Works the same way as the ucreate relay modules.
Relay 9 is the same, regardless of the number of relays.

thunkableBob, 2020-09-01:
Thanks for that, Darryl.
This problem shows on a laptop using Linux Mint 19.3. Even in a command
line, the relays activate individually.
When I run it on Ubuntu 20.04 in Python 3, if I put all in a command line,
then those relays activate at the same time.

darrylb123, 2020-09-01:
So you are using the python library. That's cool, the relay 9 change should
work fine in python too.

closed 2020-09-14 by darrylb123