darrylb123/usbrelay #58: Quiet output not working

hamir-suspect, 2020-09-01:
Hi, I can not get the quiet output to work 
What I tried is following, whatever I do I always get Device found output...

```
pi@raspberrypi:~ $ sudo usbrelay
Device Found
  type: 16c0 05df
  path: /dev/hidraw2
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay2
  Release:      100
  Interface:    0
  Number of Relays = 2
1_1=1
1_2=1

pi@raspberrypi:~ $ sudo usbrelay -q
Orig: -q, Serial: -q, Relay: 0 State: 0
Device Found
  type: 16c0 05df
  path: /dev/hidraw2
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay2
  Release:      100
  Interface:    0
  Number of Relays = 2
Serial: -q, Relay: 0 State: 0 

Serial: -q, Relay: 0 State: 0 --- Not Found

```

darrylb123, 2020-09-01:
You are not running a later version. You will need to compile from the
GitHub repository. Please follow the instructions to compile.

hamir-suspect, 2020-09-03:
Thanks, all good.

closed 2020-09-03 by hamir-suspect

hamir-suspect, 2020-09-05:
I have one more question, do you have experience with having multiple boards plugged in, I am wondering if they all really have unique serial number, and if not how can you differentiate between them if you do not use path but serial number

darrylb123, 2020-09-06:
Hi, I have several boards that I test together. Each board has it's own
serial. The serial can be changed if they are not unique. You can also mix
dcctech boards with ucreate boards. You can also differentiate by device.
How to do this is in the Readme on GitHub.

Darryl