darrylb123/usbrelay #59: Memory error

kamil-git-lab, 2020-09-07:
When I run usbrelay command , it gives following memory error,
$ sudo usbrelay
free(): invalid next size (fast)
Aborted

darrylb123, 2020-09-07:
Please run the command with -d to confirm the executable and library is in
sync.

kcgitlb, 2020-09-07:
Still gives same error (at last line), and can't change state

> $ sudo usbrelay -d
> Found 1 devices
> Device Found
>  type: 16c0 05df
>   path: /dev/hidraw0
>   serial_number: KC1MB

> 
> Manufacturer: www.dcttech.com
>   Product:      USBRelay1
>   Release:      100
>   Interface:    0
>   Number of Relays = 1
>   Module_type = 1
> KC1MB_1=0
> Device Found
>   type: 16c0 05df
>   path: /dev/hidraw1
>   serial_number: KC1PW
> Manufacturer: www.dcttech.com
>   Product:      USBRelay1
>   Release:      100
>   Interface:    
> 
> 
>   Number of Relays = 1
>   Module_type = 1
> KC1PW_1=0
> **free(): invalid next size (fast)
> Aborted**

kcgitlb, 2020-09-07:
It's fixed when connected relays to different ports.

darrylb123, 2020-09-07:
Looks like an old version too. Can you clone and build the latest version
and try again on both the dodgy port and a working port. See the readme for
how to build it from source.

Darryl

closed 2020-09-08 by kcgitlb