darrylb123/usbrelay #61: Device renamed to empty name

BigBrower, 2021-01-01:
Hi,
apparently my device has gotten an empty name, and now I cant use it nor rename it anymore.
I am on Raspian OS and just installed the package via apt-get.
Before I had renamed the device to "rel01", then I had called "usbrelay rel01", which apparently somehow set an empty device name.

> pi@raspidb:~ $ usbrelay
> Device Found
>   type: 16c0 05df
>   path: /dev/hidraw0
>   serial_number:
>   Manufacturer: www.dcttech.com
>   Product:      USBRelay8
>   Release:      100
>   Interface:    0
>   Number of Relays = 8
> rel01_1=0
> rel01_2=0
> rel01_3=0
> rel01_4=0
> rel01_5=0
> rel01_6=0
> rel01_7=0
> rel01_8=0
> 
> pi@raspidb:~ $ usbrelay rel01
> Orig: rel01, Serial: rel01, Relay: 0 State: 0
> Device Found
>   type: 16c0 05df
>   path: /dev/hidraw0
>   serial_number:
>   Manufacturer: www.dcttech.com
>   Product:      USBRelay8
>   Release:      100
>   Interface:    0
>   Number of Relays = 8
> Serial: rel01, Relay: 0 State: 0
> 1 HID Serial: rel01
> Serial: rel01, Relay: 0 State: 0 --- Not Found
> pi@raspidb:~ $ usbrelay
> Device Found
>   type: 16c0 05df
>   path: /dev/hidraw0
>   serial_number:
>   Manufacturer: www.dcttech.com
>   Product:      USBRelay8
>   Release:      100
>   Interface:    0
>   Number of Relays = 8
> _1=0
> _2=0
> _3=0
> _4=0
> _5=0
> _6=0
> _7=0
> _8=0

darrylb123, 2021-01-01:
The version installed from apt-get is very old. Please compile the latest
version from GitHub. The new version has the ability to refer to it by its
device name and can be renamed using that.
Just follow the build instructions, it's very straight forward.

BigBrower, 2021-01-02:
Thanks for the very quick answer. With the new version I was able to rename it with
`sudo usbrelay _0=XX`
and everything is working fine again. Was already wondering why some stuff seems a little different compared to the readme. The Issue can be closed then.
Maybe a little clue in the readme that the apt-get package is outdated would be helpful?

Btw, according to the help, you can get the version with "-v", but actually you have to put in two "--":
sudo usbrelay --help

> Usage: usbrelay [OPTION...] [ACTION...]
> Control or query USB HID relays.
> 
>   -d, --debug                Produce debugging output
>   -q, --quiet                Be quiet
>   -?, --help                 Give this help list
>       --usage                Give a short usage message
>   -V, --version              Print program version
> 
> Without ACTION, the actual state of all relays is printed to stdout.
> ACTION can be one of:
> RELID_N=[0|1] to switch the N-th relay off or on
> RELID=NEWID to change relay ID
> 
> Report bugs to https://github.com/darrylb123/usbrelay/issues.
> pi@raspidb:~/usbrelay $ sudo usbrelay -v
> usbrelay: invalid option -- 'v'
> Try `usbrelay --help' or `usbrelay --usage' for more information.
> pi@raspidb:~/usbrelay $ sudo usbrelay --v
> usbrelay 0.7-37-gee75dd5969
>

wentasah, 2021-01-02:
> Btw, according to the help, you can get the version with "-v", but actually you have to put in two "--":
> sudo usbrelay --help

The single dash option is capital V. Double dash option is just an abbreviation of `--version`. If there was `--verbose` option, `--v` would not work, because it would not be unambiguous, but `--vers` would work.

closed 2021-01-03 by BigBrower