darrylb123/usbrelay #64: Serial changes (despite sudo)

BigBrower, 2021-04-11:
Hi,
First of all thank you for usbrelay!
I am having the Problem that the serial changes every now and then to some strange string. I couldnt see a pattern.
I am triggering Usbrelay from a home automation Software (fhem) and am using sudo always..
Any thoughts?

darrylb123, 2021-04-11:
Sorry, no idea.
Why not write a little wrapper script that logs the commands. Call that
instead.
Something like

#!
echo $@ >>/var/tmp/usbrelay.log
usbrelay $#

Then check the log next time it changes.

BigBrower, 2021-04-11:
I actually have a log, but yeah I just improved it, so that I should have all information that i need. Will see if I can find the problem this way.
Thanks so far :)

BigBrower, 2021-04-16:
Hi,
I have different scenarios, one of them is the following, and maybe you have seen something like that before.
When calling sudo usbrelay (without parameter) the following error message came up and after that one relayboard was no longer found (I had to switch everything off and on again):
```
ioctl (GFEATURE): Broken pipe
hid_get_feature_report
: Invalid argument
```

See the following log for the complete scenario:

```
2021.04.04 18:10:23 1: USBRelay_Set:name:USBRelay_rel01_2_Rollo_Eltern_Fenster_Rechts_down, cmd:off devname rel01_2
2021.04.04 18:10:23 1: USBRelay_Set: switch relay off.
2021.04.04 18:10:23 1: sudo_usbrelay: execute command: sudo usbrelay rel01_2=0
2021.04.04 18:10:23 1: sudo_usbrelay: command returned: rel02_1=0
rel02_2=0
rel02_3=0
rel02_4=0
rel02_5=0
rel02_6=0
rel02_7=0
rel02_8=0
rel01_1=0
rel01_2=1
rel01_3=0
rel01_4=0
rel01_5=0
rel01_6=0
rel01_7=0
rel01_8=0
rel03_1=0
rel03_2=0
rel03_3=0
rel03_4=0
rel03_5=0
rel03_6=0
rel03_7=0
rel03_8=0
rel04_1=0
rel04_2=0
rel04_3=0
rel04_4=0
rel04_5=0
rel04_6=0
rel04_7=0
rel04_8=0

2021.04.04 18:10:23 1: sudo_usbrelay: execute command: sudo usbrelay 
ioctl (GFEATURE): Broken pipe
hid_get_feature_report
: Invalid argument
2021.04.04 18:10:23 1: sudo_usbrelay: command returned: rel02_1=0
rel02_2=0
rel02_3=0
rel02_4=0
rel02_5=0
rel02_6=0
rel02_7=0
rel02_8=0
rel03_1=0
rel03_2=0
rel03_3=0
rel03_4=0
rel03_5=0
rel03_6=0
rel03_7=0
rel03_8=0
rel04_1=0
rel04_2=0
rel04_3=0
rel04_4=0
rel04_5=0
rel04_6=0
rel04_7=0
rel04_8=0

2021.04.04 18:10:24 1: USBRelay_Set:name:USBRelay_rel01_1_Rollo_Eltern_Fenster_Links_down, cmd:on devname rel01_1
2021.04.04 18:10:24 1: USBRelay_Set: switch relay on.
2021.04.04 18:10:24 1: sudo_usbrelay: execute command: sudo usbrelay rel01_1=1
2021.04.04 18:10:28 1: sudo_usbrelay: command returned: rel02_1=0
rel02_2=0
rel02_3=0
rel02_4=0
rel02_5=0
rel02_6=0
rel02_7=0
rel02_8=0
rel03_1=0
rel03_2=0
rel03_3=0
rel03_4=0
rel03_5=0
rel03_6=0
rel03_7=0
rel03_8=0
rel04_1=0
rel04_2=0
rel04_3=0
rel04_4=0
rel04_5=0
rel04_6=0
rel04_7=0
rel04_8=0
```

Is it maybe a problem that the command was called so quick after eachother?
In the last call I also noticed that it took 4 seconds until the response came.

Any idea? Seen anything similar before?

Thanks in advance!

darrylb123, 2021-04-16:
Are you running the latest version from GitHub? If not, please build and
run the test again with that version.
- If it fails again, can you run dmesg and see if the USB device
disconnected.
- Run lsusb before the test and compare with after it fails.
- Disconnect the failed relay and see if any others exhibit the same issue.
- Try swapping cables and ports and retest

I'm thinking it's faulty hardware, just not sure which.

Darryl