darrylb123/usbrelay #65: usbrelay stops giving any output

ESPWarren, 2021-05-05:
I have USBrelay hosted on a Pi. It works well and has 2 relays. I have renamned my two relays as WATER_1 and WATER_2

If I type simply "sudo usbrelay" then the output is:
WATER_1=0
WATER_2=0

However, occasionally it stops providing output at all - and lsusb shows that the USB device itself is no longer a visible USB device - it is as if it is no longer  plugged in.

The only fix I have so far found is to reboot.

Do you know any way to reset that particular USB device without rebooting or disturbing the USB hub itself?

Thanks...

wentasah, 2021-05-05:
Do you see any interesting message in the output of `dmesg`? I've seen many similar-looking USB-related errors caused by bad cables or unstable/overloaded power supply. You seem to use USB hub. Powering the hub with independent power supply may help.

You can reset a USB device with the `usbreset` command from the [usbutils](https://github.com/gregkh/usbutils/blob/master/usbreset.c) package.

darrylb123, 2021-05-05:
I suspect the underscore is confusing it. Can you rename and remove the
underscore from the name. Use a hyphen (-) instead.
It would find the first underscore and look for a module called WATER

darrylb123, 2021-05-05:
Sorry, ignore my first response. Might help if I read your email properly.
Michal has the right advice.

closed 2021-09-01 by darrylb123