darrylb123/usbrelay #81: Module usbrelay_py has no attribute 'board_control'

joshuar87, 2022-03-09:
Hi,

usbrelay seems to be installed in ubuntu (mint) fine. It imports ok into python but none of the attributes seem to be there. I ran make and install for python.

In the meantime going to use a workaround:
import os
os.system("usbrelay...

caveman2st, 2022-03-17:
i have the same issues:

>>> import usbrelay_py
>>> dir(usbrelay_py)
['__author__', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__', '__version__']

darrylb123, 2022-03-17:
I can confirm the error on Linux Mint 20.3
I am neither a python developer nor a ubuntu/debian user so I am in the dark here.
The issue is not on Fedora so it must be solvable.
I will need to find someone that can help.

darrylb123, 2022-03-17:
Ok, I have no idea what changed, but in the process of preparing usbrelay to become a Fedora package the python build system was changed over to the modern method.
I tested the build using the new code and it works file. I did have to make a change to the Makefile so you will need the absolute latest.
To try it, pull a fresh version from github (git pull) then check out the new branch.
$ git checkout python_wheel

There is an updated README and additional packages for python package building will need to be installed.
# sudo apt install libpython3-dev python3-venv pip
# sudo pip install build

You may also need to delete the usbrelay packages installed under /usr/local/lib/python3.X/site-packages before installing

Keep a record of anything different from the doco you had to do and I will update the docs. I'm very unfamiliar with a debian environment.

joshuar87, 2022-03-17:
Thanks a lot for the prompt reply will give it a try soon!

caveman2st, 2022-03-17:
Now it works for me on Debian.
Thank You

python library built and install works different:
cd usbrelay_py
make

darrylb123, 2022-03-17:
Yes, the new release coming builds a python wheel package.

closed 2022-05-16 by darrylb123