darrylb123/usbrelay #84: Ok with pi 1 B but not anymore with pi 4 B

gortador, 2022-04-07:
Hello, I realized that the relay was detected by lsusb in the raspberry pi 1 B but was no longer detected with the pi 4 B, however both are up to date, could you tell me how I can make it recognize? what libraries or manipulatio to do please

gortador, 2022-04-07:
> `lsusb` does not rely on any library related to relays. If the Linux kernel sees the relay, `lsusb` should show it. If not, it is usually a hardware (cable) problem. Do you see some suspicious messages in the `dmesg` output? Usually something like "usb enumeration failed" or "usb device timeout".

I see absolutely nothing, as if the relay was not connected, on the other hand it appears well on my pi 1 B so it works and is recognized but not on my pi 4 B

wentasah, 2022-04-07:
Does any other USB device work (is visible in lsusb) on 4B? `dmesg` should show at least something like:

    usb 2-3: new SuperSpeed USB device number 10 using xhci_hcd

gortador, 2022-04-07:
yes all my other peripherals work, and this even if I change the sockets, the relay is not displayed anywhere of the 4 usb ports of the pi4, but is clearly visible on all the ports of the pi1

wentasah, 2022-04-07:
That's strange. I've never seen this. When connecting a relay to my Pi 4 B, I see this:
```
[16587.681067] usb 1-1.3: new low-speed USB device number 8 using xhci_hcd
[16587.788880] usb 1-1.3: New USB device found, idVendor=16c0, idProduct=05df, bcdDevice= 1.00
[16587.788898] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[16587.788910] usb 1-1.3: Product: USBRelay4
[16587.788921] usb 1-1.3: Manufacturer: www.dcttech.com
[16587.796939] hid-generic 0003:16C0:05DF.0001: hiddev96,hidraw0: USB HID v1.01 Device [www.dcttech.com USBRelay4] on usb-0000:01:00.0-1.3/input0
[16587.830135] usbcore: registered new interface driver radio-ma901
```
lsusb shows this:
```
Bus 002 Device 006: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 002 Device 005: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 002 Device 004: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 002 Device 003: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. Hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 008: ID 16c0:05df Van Ooijen Technische Informatica HID device except mice, keyboards, and joysticks
Bus 001 Device 007: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 001 Device 006: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 001 Device 005: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 001 Device 004: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 001 Device 003: ID 0bda:5411 Realtek Semiconductor Corp. RTS5411 Hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
I guess you've already tried different cables.

darrylb123, 2022-05-16:
No response, I assume new cables fixed it.

closed 2022-05-16 by darrylb123