darrylb123/usbrelay #90: Change relay ID issue "_1="

younisamedi, 2022-08-09:
Hello, I attempted changing the relay ID or my board but now I no longer see any ID names. It completely removed the name and when I try to list the relays with "sudo usbrelay" I only see:

_1=0
_2=0
_3=0
_4=0
_5=0
_6=0
_7=0
_8=0

I can't turn them on/off without the name.

#################################

# The command I used to change the name: 

sudo usbrelay HW554=PLUG0

# See the change result:

Device Found
  type: 16c0 05df
  path: /dev/hidraw3
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
_1=0
_2=0
_3=0
_4=0
_5=0
_6=0
_7=0
_8=0

#################################

I have also tried this but didn't work:

# command:
usbrelay /dev/hidraw3_0=JJII3

# output:
*** buffer overflow detected ***: terminated
Aborted

#################################

I tried this on two different boards (both 8 relays), but still no luck.
I hope to find a fix fro this, and any advise will be highly appreciated. Thank you for your time! 

- YA

younisamedi, 2022-08-09:
### Update: fixed

I originally installed the "usbrelay" from the a Debian 11 repo.

I removed usbrelay, and downloaded the latest version of usbrelay and compiled it from the source code. 

# Issue: 
sudo usbrelay

_1=0
_2=0
_3=0
_4=0
_5=0
_6=0
_7=0
_8=0

# This is how I fixed it:
sudo usbrelay _=<NEW ID NAME>    # E.g:  sudo usbrelay _=PLUG1

PLUG1_1=0
PLUG1_2=0
PLUG1_3=0
PLUG1_4=0
PLUG1_5=0
PLUG1_6=0
PLUG1_7=0
PLUG1_8=0

I think the ID name has to be 5 or more characters, don't give it a short name. 

Hope this helps. 

##############

closed 2022-08-09 by younisamedi

darrylb123, 2022-08-09:
The US relay package in debian is ancient. It does not support some of the
features you have used.
Please build it from source from the GitHub repository.
There is work happening to get the debian packages updated.

Darryl

younisamedi, 2022-08-09:
Thanks Darryl, I appreciate your work.