darrylb123/usbrelay #93: Using the Python Extensions in Windows

williamblair333, 2022-09-21:
I'm certain I'm being ignorant - but I had to ask:  Can the python extensions of usbrelay be exported and used in Windows?  Apologies for posting here as I didn't know where else to ask.  Thanks.

wentasah, 2022-09-22:
> I'm certain I'm being ignorant - but I had to ask: Can the python
> extensions of usbrelay be exported and used in Windows?

I guess it's theoretically possible, but it will need some work. We use
the hidapi library, which should work on Windows, but I don't think that
our code has ever be compiled for Windows. We probably have few Linux
specific things in our code and if somebody replaces them with platform
independent code, it might work.

If time allows, I might look into this eventually.

darrylb123, 2022-09-22:
I had a quick look at mingw and cross compiling, looks like it's doable for
the command line tool and library. The python extension is another matter.

williamblair333, 2022-09-27:
I may not be useful, but, what can someone do to assist?

darrylb123, 2022-10-11:
No. The python extension is a wrapper around libusbrelay.

I need to find someone that uses/develops on windows to add the required
code for windows.

Darryl

closed 2024-05-01 by williamblair333