darrylb123/usbrelay #96: Fedora repository

aldogig, 2022-12-03:
In the initial page you say that for Fedora it is possible to issue the command:
_**$ sudo dnf install usbrelay-common python3-usbrelay usbrelay-mqtt**_
however I get non result. 
I believe a repository shall be first added.
Can you state, as well, in which repository can be found the above packages?
Many thanks 
Aldo

darrylb123, 2022-12-03:
Yes, we are still working on getting the packages in the Fedora
repositories.
They are currently under review. The test build packages are here.
https://copr.fedorainfracloud.org/coprs/fuller/usbrelay/build/5073353/

Darryl

darrylb123, 2023-01-09:
Currently in updates-testing. Should be in main repository in a day or 2

closed 2023-01-09 by darrylb123

aldogig, 2023-01-09:
Many thanks.