darrylb123/usbrelay #99: USBRelay device naming bug

wasikan-git, 2023-01-16:
Hi,
I think there's a bug in the code.
I've a relay which identifies itself (sudo usbrelay -e /dev/hidraw4) as:

ELSD__1=0
ID_SERIAL=ELSD_

Please note underscore as a last character of ID_SERIAL.

sudo usbrelay -d gives:

libusbrelay: 1.0-2~bpo11+1
usbrelay: 1.0-2~bpo11+1
enumerate_relay_boards()Found 1 devices
Device Found
  type: 16c0 05df
  path: /dev/hidraw4
  serial_number: ELSD_
Manufacturer: www.dcttech.com
  Product:      USBRelay1
  Release:      100
  Interface:    0
  Number of Relays = 1
  Module_type = 1
ELSD__1=0

and sudo usbrelay ELSD__1=1
Invalid relay specification: ELSD__1=1

I also have another relay of the same type but its ID_SERIAL does not end with underscore. This relay works as expected.
Hopefully usbrelay can be fixed without too much trouble.

Regards
Andrzej

darrylb123, 2023-01-16:
Yep, looks like a bug.
In the mean time, why not change the one with the trailing underscore to
remove it. You can use the device path to change the name.
Find out the device path for your relay, it seems to be the same as the
documentation (/dev/hidraw4) . Use usbrelay -v to find the device path
From the documentation:
$ sudo usbrelay
ZXCV_1=0
ZXCV_2=0

$ sudo usbrelay ZXCV_0=ZAQ12 # or /dev/hidraw4_0=ZAQ12
ZXCV_1=0
ZXCV_2=0
Setting new serial

$ sudo usbrelay
ZAQ12_1=0
ZAQ12_2=0

Darryl

wasikan-git, 2023-01-16:
Hi!

I followed your advice and changed the device name. Now the relay works 
as expected. Thanks.

Regards

Andrzej

W dniu 17.01.2023 o 00:24, Darryl Bond pisze:

> Yep, looks like a bug.
> In the mean time, why not change the one with the trailing underscore to
> remove it. You can use the device path to change the name.
> Find out the device path for your relay, it seems to be the same as the
> documentation (/dev/hidraw4) . Use usbrelay -v to find the device path
> From the documentation:
> $ sudo usbrelay
> ZXCV_1=0
> ZXCV_2=0
>
> $ sudo usbrelay ZXCV_0=ZAQ12 # or /dev/hidraw4_0=ZAQ12
> ZXCV_1=0
> ZXCV_2=0
> Setting new serial
>
> $ sudo usbrelay
> ZAQ12_1=0
> ZAQ12_2=0
>
> Darryl
>
>
>
> > Hi,
> > I think there's a bug in the code.
> > I've a relay which identifies itself (sudo usbrelay -e /dev/hidraw4) as:
> >
> > ELSD__1=0
> > ID_SERIAL=ELSD_
> >
> > Please note underscore as a last character of ID_SERIAL.
> >
> > sudo usbrelay -d gives:
> >
> > libusbrelay: 1.0-2bpo11+1
> > usbrelay: 1.0-2bpo11+1
> > enumerate_relay_boards()Found 1 devices
> > Device Found
> > type: 16c0 05df
> > path: /dev/hidraw4
> > serial_number: ELSD_
> > Manufacturer: www.dcttech.com
> > Product: USBRelay1
> > Release: 100
> > Interface: 0
> > Number of Relays = 1
> > Module_type = 1
> > ELSD__1=0
> >
> > and sudo usbrelay ELSD__1=1
> > Invalid relay specification: ELSD__1=1
> >
> > I also have another relay of the same type but its ID_SERIAL does 
> not end
> > with underscore. This relay works as expected.
> > Hopefully usbrelay can be fixed without too much trouble.
> >
> > Regards
> > Andrzej
-- 
Pozdrawiam
Andrzej Wasik

darrylb123, 2023-01-18:
Not a bug.
The code only allows A-Z0-9
Updated the documentation to describe how to rename in the event of illegal characters in the serial ID

closed 2023-01-18 by darrylb123