darrylb123/usbrelay #100: Lost device // HID device still there

cm0n, 2023-02-28:
Hello all,
I have a problem after some time / switching cycles not always the same.
But in the region of 10 to 20 switching. I lose the devices connection.
I get this message:
```
root@ubuntu:~# usbrelay 
ioctl (GFEATURE): Protocol error
hid_get_feature_report
: Protocol error
```

The HID device but is visible, I have also tried it with older version of usbrelay, same behavior.
```
root@ubuntu:~# lsusb 
Bus 003 Device 002: ID 8087:8000 Intel Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 028: ID 16c0:05df Van Ooijen Technische Informatica HID device except mice, keyboards, and joysticks
Bus 001 Device 003: ID 1c4f:0034 SiGma Micro 
Bus 001 Device 002: ID 1c4f:0002 SiGma Micro Keyboard TRACER Gamma Ivory
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

What more do you need to help me?
I had before this realy with  [4 ports](https://www.amazon.de/Relaismodul-4-Kanal-Computer-Controller-Erweiterungskarte-default/dp/B08KS6G581/ref=sr_1_4?keywords=usbrelay&link_code=qs&qid=1677574039&sourceid=Mozilla-search&sr=8-4) now I have it with [8 ports](https://www.amazon.de/8-Kanal-Computer-Controller-importiertem-Hochleistungs-USB/dp/B07L12DXWB/ref=d_pd_sbs_sccl_2_6/258-5057675-0666766?pd_rd_w=cz0Y0&content-id=amzn1.sym.6023e427-25bb-42e7-aaa9-a6e0a8bdcb7b&pf_rd_p=6023e427-25bb-42e7-aaa9-a6e0a8bdcb7b&pf_rd_r=TJTHDC2WM5NAH8586FYG&pd_rd_wg=PIdfj&pd_rd_r=af613a2c-b88a-4db2-a026-959fc844f833&pd_rd_i=B07L12DXWB&psc=1) and it makes problems the one with 4 runs without problems, but in a other machine.

```
root@ubuntu:~# usbrelay -d
libusbrelay: 1.1.2-8-g88844b073d-dirty
usbrelay: 1.1.2-8-g88844b073d-dirty
enumerate_relay_boards()Found 0 devices
root@ubuntu:/home/aaronia# usbrelay -e 1
```
I have to disconnect it completely from the power. 
Since this is currently from my wiring only possible when I turn off the computer and pull the power plug. After that it runs again for a short time.

many greetings

darrylb123, 2023-02-28:
I suspect hardware somewhere.
Can you try a different cable, port etc.

Darryl

cm0n, 2023-02-28:
Yes I have already used both libs, same behavior.
I have also tried all 4 USB ports.
Also all 3 relay cards I have show the same behavior.

So is this rather an error from the card itself?

In a previous project with the version with 4 I had no problems.
Is there any way to debug it more to find out why it happens?

cm0n, 2023-02-28:
Some more information I could collect, maybe it can help?

When it is working fine:
```
root@ubuntu:~$ usbhid-dump
001:010:000:DESCRIPTOR         1677585220.646253
 06 00 FF 09 01 A1 01 15 00 26 FF 00 75 08 95 08
 09 00 B2 02 01 C0

001:004:000:DESCRIPTOR         1677585220.648131
 05 01 09 02 A1 01 09 01 A1 00 05 09 19 01 29 03
 15 00 25 01 95 03 75 01 81 02 95 01 75 05 81 01
 05 01 09 30 09 31 09 38 15 81 25 7F 75 08 95 03
 81 06 C0 C0

001:005:001:DESCRIPTOR         1677585220.650114
 05 0C 09 01 A1 01 85 01 19 00 2A 3C 02 15 00 26
 3C 02 95 01 75 10 81 00 C0 05 01 09 80 A1 01 85
 02 19 81 29 83 25 01 75 01 95 03 81 02 95 05 81
 01 C0

001:005:000:DESCRIPTOR         1677585220.652043
 05 01 09 06 A1 01 05 08 19 01 29 03 15 00 25 01
 75 01 95 03 91 02 95 05 91 01 05 07 19 E0 29 E7
 95 08 81 02 75 08 95 01 81 01 19 00 29 91 26 FF
 00 95 06 81 00 C0
```

```
root@ubuntu:~$ usb-devices
....

T:  Bus=01 Lev=01 Prnt=01 Port=05 Cnt=04 Dev#=  2 Spd=1.5 MxCh= 0
D:  Ver= 1.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=16c0 ProdID=05df Rev=01.00
S:  Manufacturer=www.dcttech.com
S:  Product=USBRelay8
C:  #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=20mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=usbhid

....
```
when it stops working:

```
root@ubuntu:~$ usbhid-dump
001:014:000:Failed to retrieve report descriptor: Input/Output Error
001:014:000:Failed to attach to the kernel driver: Entity not found
```

```
root@ubuntu:~$ usb-devices
....
T:  Bus=01 Lev=01 Prnt=01 Port=05 Cnt=04 Dev#= 14 Spd=1.5 MxCh= 0
D:  Ver= 1.10 Cls=00(>ifc ) Sub=00 Prot=00 MxPS= 8 #Cfgs=  1
P:  Vendor=16c0 ProdID=05df Rev=01.00
S:  Manufacturer=www.dcttech.com
S:  Product=USBRelay8
C:  #Ifs= 1 Cfg#= 1 Atr=80 MxPwr=20mA
I:  If#= 0 Alt= 0 #EPs= 1 Cls=03(HID  ) Sub=00 Prot=00 Driver=(none)

.....
```

As what I can see, the drive breaks somewhere?!

BR

darrylb123, 2023-02-28:
That indicates it is something in the hid stack if is not just affecting
usbrelay.
I use Fedora and raspian without issues, perhaps try a VM with newer/older
version of Ubuntu or Fedora??

darrylb123, 2023-03-01:
I just tested a relay switching from off to on  100 times without a problem on Fedora 37.

darrylb123, 2023-03-01:
> Hi,
> I have now tried it on a ubuntu 18.04. The first system was an ubuntu
> 14.04.
> Unfortunately the same behavior.
> No matter against which lib I compile.
>
> Is there a way to debug it better?
> Is this a board error or is it somewhere else?
> All three of the boards that I have show the same misbehaviour,
> unfortunately.

Is it the same computer?
I'm not sure how to help from here.

durian, 2023-09-28:
I know, old topic, but I seem to have the same problem. Sometimes the relay-card (8 ports) disappears, and I have to reboot the computer to get it back.

This is on a raspberry pi.
`uname -a
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux

usbrelay -d
libusbrelay: 1.2.1
usbrelay: 1.2.1
enumerate_relay_boards()Found 1 devices
Device Found
  type: 16c0 05df
  path: /dev/hidraw0
  serial_number: REL0
Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
  Module_type = 1
REL0_1=0
REL0_2=0
REL0_3=0
REL0_4=0
REL0_5=0
REL0_6=0
REL0_7=0
REL0_8=0
`

darrylb123, 2023-09-28:
Nothing really to do with usbrelay, but I'd check the dmesg log to see if
it registered as removed sometime previously, also lsusb to see if the
kernel can see it.
Does an unplug/replug also fix it?
Most times I have come across this, it's a dodgy cable.

durian, 2023-09-29:
> Nothing really to do with usbrelay, but I'd check the dmesg log to see if it registered as removed sometime previously, also lsusb to see if the kernel can see it. Does an unplug/replug also fix it? Most times I have come across this, it's a dodgy cable.

Thanks, I'll see if I can catch it again, and find out what happens. I couldn't try unplugging the cable as the raspberry pi is on a remote location. Rebooting fixed the problem though.

liebowa, 2024-09-04:
I know this is a a very old thread, but I face this issue also.  I've changed cables, etc.  When this happens, I've tried to unplug/replug to no avail.  I installed an USBResetter app and tried all of it's options (reset just the device, reset the entire hub, etc.), but it does not come back (show back up as a device) until a full reboot.

I'm 100% certain this is not an issue with this software, but also trying to find out from anyone if there are any other thoughts on how to reset the USB hub.

Thanks in advance.