darrylb123/usbrelay #101: Not able to run without sudo even after using udev rules

gits00, 2023-03-15:
Tried - 

$ sudo cp 50-usbrelay.rules /etc/udev/rules.d
$ sudo udevadm control -R
The command after that - 
sudo usermod -a -G usbrelay <user name>
didn't work.
Gives error - usermod: group 'usbrelay' does not exist
How to add the usbrelay group?

Output -

python3 usbrelay_test.py 

/dev/hidraw4: Permission denied
Unable to open any device - Use root, sudo or set the device permissions via udev
Count:  1
Boards:  (('', 8, 0),)
Board:  ('', 8, 0)
operate_relay() Unable to write or unknown relay
Error: (null)
Result:  -1
operate_relay() Unable to write or unknown relay
Error: (null)
Result:  -1
operate_relay() Unable to write or unknown relay
Error: (null)
Result:  -1
operate_relay() Unable to write or unknown relay
Error: (null)
Result:  -1
operate_relay() Unable to write or unknown relay
Error: (null)
Result:  -1
operate_relay() Unable to write or unknown relay
Error: (null)
Result:  -1
operate_relay() Unable to write or unknown relay
Error: (null)
Result:  -1
operate_relay() Unable to write or unknown relay
Error: (null)
Result:  -1
operate_relay() Unable to write or unknown relay
Error: (null)
Result:  -1
operate_relay() Unable to write or unknown relay
Error: (null)
Result:  -1
operate_relay() Unable to write or unknown relay
Error: (null)
Result:  -1
operate_relay() Unable to write or unknown relay
Error: (null)
Result:  -1
operate_relay() Unable to write or unknown relay
Error: (null)
Result:  -1
operate_relay() Unable to write or unknown relay
Error: (null)
Result:  -1
operate_relay() Unable to write or unknown relay
Error: (null)
Result:  -1
operate_relay() Unable to write or unknown relay
Error: (null)
Result:  -1

 usbrelay 
/dev/hidraw4: Permission denied
Unable to open any device - Use root, sudo or set the device permissions via udev

I am using Ubuntu 18.04.5 LTS

darrylb123, 2023-03-15:
If you are using Ubuntu or debian substitute the group plugdev for usbrelay.

Darryl

darrylb123, 2023-03-17:
Note I've updated the README to show this.

closed 2023-03-17 by darrylb123

gits00, 2023-03-17:
Thanks! This worked.