darrylb123/usbrelay #102: Weird status of power relay

Shihao-Feng, 2023-03-20:
I have bought a 4-way relay half an year ago and it could use with usbrelay properly at that time. However, when I took it out this morning, tried it with simple commands, it give weird output:
```
shihaofeng@shihaofeng-U-65LKN42:~/Downloads/usbrelay-1.0.1$ sudo usbrelay
Device Found
  type: 16c0 05df
  path: /dev/hidraw5
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay4
  Release:      100
  Interface:    0
  Number of Relays = 4
�AAMZ_1=0
�AAMZ_2=0
�AAMZ_3=0
�AAMZ_4=0

shihaofeng@shihaofeng-U-65LKN42:~/Downloads/usbrelay-1.0.1$ sudo usbrelay /dev/hidraw5_0=0
*** buffer overflow detected ***: terminated
Aborted

```

Is it a software issue or hardware issue? Just want to confirm it. If it is a hardware one, I would just bought a new one.

darrylb123, 2023-03-20:
Looks like the serial was corrupted.
Try changing its name

sudo usbrelay /dev/hidraw5_0=AAAMZ

You tried to change its name to 0

Shihao-Feng, 2023-03-20:
> Looks like the serial was corrupted. Try changing its name sudo usbrelay /dev/hidraw5_0=AAAMZ You tried to change its name to 0
> […](#)

Just tried
```
shihaofeng@shihaofeng-U-65LKN42:~/Downloads/usbrelay-1.0.1$ sudo usbrelay /dev/hidraw5_0=AAAMZ
*** buffer overflow detected ***: terminated
Aborted

```

darrylb123, 2023-03-21:
Just ran a similar test on a dual relay. Note the relay serial is not invalid though. 
$ sudo usbrelay -d
libusbrelay: 1.1.2-5-g67df72677c-dirty
usbrelay: 1.1.2-5-g67df72677c-dirty
enumerate_relay_boards()Found 1 devices
Device Found
  type: 16c0 05df
  path: /dev/hidraw4
  serial_number: OMG12
Manufacturer: www.dcttech.com
  Product:      USBRelay2
  Release:      100
  Interface:    0
  Number of Relays = 2
  Module_type = 1
OMG12_1=0
OMG12_2=0
$
$ sudo usbrelay -d /dev/hidraw4_0=ZXCVB
Orig: /dev/hidraw4_0=ZXCVB, Serial: /dev/hidraw4, Relay: 0 State: 0
libusbrelay: 1.1.2-5-g67df72677c-dirty
usbrelay: 1.1.2-5-g67df72677c-dirty
enumerate_relay_boards()Found 1 devices
Device Found
  type: 16c0 05df
  path: /dev/hidraw4
  serial_number: OMG12
Manufacturer: www.dcttech.com
  Product:      USBRelay2
  Release:      100
  Interface:    0
  Number of Relays = 2
  Module_type = 1
OMG12_1=0
OMG12_2=0
main() arg 0 Serial: /dev/hidraw4, Relay: 0 State: 0 
find_board(/dev/hidraw4) path /dev/hidraw4
Setting new serial
find_board(OMG12) path /dev/hidraw4
$
$ sudo usbrelay -d
libusbrelay: 1.1.2-5-g67df72677c-dirty
usbrelay: 1.1.2-5-g67df72677c-dirty
enumerate_relay_boards()Found 1 devices
Device Found
  type: 16c0 05df
  path: /dev/hidraw4
  serial_number: ZXCVB
Manufacturer: www.dcttech.com
  Product:      USBRelay2
  Release:      100
  Interface:    0
  Number of Relays = 2
  Module_type = 1
ZXCVB_1=0
ZXCVB_2=0

darrylb123, 2023-03-21:
Did you build it from the latest source from github?

darrylb123, 2023-03-21:
> no luck with the any version, I switched to use the hid interface to fix
> the serial name, now it is working again.

What do you mean, you switched to the hid interface?
Just want the fix detailed if the same problem comes up again.

Shihao-Feng, 2023-03-21:
> What do you mean, you switched to the hid interface? Just want the fix detailed if the same problem comes up again.
> […](#)

I am not sure where the problem is, all version returns buffer overflow detected, so I just use the hid_write directly, and it do fix the issue.

darrylb123, 2023-03-21:
Can you post the command line you used to fix it?

Can you use usbrelay now?

Shihao-Feng, 2023-03-21:
> Can you post the command line you used to fix it? Can you use usbrelay now?
> […](#)

I can use it now but its output is still weird
```
shihaofeng@shihaofeng-U-65LKN42:~/work$ sudo usbrelay --help
Orig: --help, Serial: --help, Relay: 0 State: 0
Device Found
  type: 16c0 05df
  path: /dev/hidraw5
  serial_number: 
  Manufacturer: www.dcttech.com
  Product:      USBRelay4
  Release:      100
  Interface:    0
  Number of Relays = 4
Serial: --help, Relay: 0 State: 0 

Serial: --help, Relay: 0 State: 0 --- Not Found
```

for the command I use to fix, I just do the similar thing you did [here](https://github.com/darrylb123/usbrelay/blob/69b245056b121ca1385fa3120741cee406e3d98d/libusbrelay.c#L261), so I don't know why it actually fix the issue but usbrelay built from any version cannot .

darrylb123, 2023-03-21:
I really don't understand. This is what is supposed to happen when you use the --help option.
$ usbrelay --help
Usage: usbrelay [OPTION...] [ACTION...]
Control or query USB HID relays.

  -d, --debug                Produce debugging output
  -e, --export-id=DEV        Print relay ID of the given DEV (/dev/hidrawXX) in
                             udev-compatible format
  -q, --quiet                Be quiet
  -?, --help                 Give this help list
      --usage                Give a short usage message
  -V, --version              Print program version

Mandatory or optional arguments to long options are also mandatory or optional
for any corresponding short options.

Without ACTION, the actual state of all relays is printed to stdout.
ACTION can be one of:
RELID_N=[0|1] to switch the N-th relay off or on
RELID_0=NEWID to change relay ID

Report bugs to https://github.com/darrylb123/usbrelay/issues.

darrylb123, 2023-03-21:
Have you every installed usbrelay from the debian package?

What do you get when you run:
$usbrelay -V
$ sudo usbrelay -d

Shihao-Feng, 2023-03-21:
> Have you every installed usbrelay from the debian package?
> 
> What do you get when you run: $ sudo usbrelay -V $ sudo usbrelay -d

I think I figured out what happen, my laptop did not download from debian directly but from a local server, but that copy is old(something like 0.4). The ones is not cleaned up throughly, the newly downloaded and built one did not replace it correctly. Thanks for your time.

Shihao-Feng, 2023-03-21:
> looks to me like you have an ancient version installed. Probably from the debian package. Can you run sudo apt remove usbrelay and rebuild and re-install it.

It works properly now. Great thanks for your patience and time!

closed 2023-03-21 by Shihao-Feng

ybyllc, 2026-11-09:
我也遇到了