darrylb123/usbrelay #103: New board 5131:2007 not working

alfem, 2023-04-17:
Hi!

I recently bought a similar 2 relays usb board. Vendor/product id are different (5131:2007).  You can read "lcus_hid_x2" on the PCB.

I replaced the original usb ids with these new ones in RelayModuleDcttech.h, but no luck. 

This is the new board:
https://es.aliexpress.com/item/1005003595023772.html

A dmesg shows this info: 
```
[17817.335415] usb 1-1: new full-speed USB device number 13 using xhci_hcd
[17817.483980] usb 1-1: New USB device found, idVendor=5131, idProduct=2007, bcdDevice= 0.00
[17817.483999] usb 1-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[17817.489547] usb 1-1: language id specifier not provided by device, defaulting to English
[17817.498063] hid-generic 0003:5131:2007.0009: hiddev2,hidraw6: USB HID v1.00 Device [HID 5131:2007] on usb-0000:00:14.0-1/input0
```
With ids changed I only get this:
```
path:             /dev/hidraw6
vendor_id:        0x5131
product_id:       0x2007
serial_number:    
release_number:   0
manufacturer:     (null)
product:          (null)
interface_number: 0
module_name:
```

Any ideas?

alfem, 2023-04-17:
I think this is a CFSunbird board, probably with a different protocol. In the official store I found some useful info about serial communication in the item description:

https://es.aliexpress.com/i/1005003781042941.html

Thanks for your help!

alfem, 2023-04-17:
Finally I tested sending hex codes right to the hidraw device file and it worked!

Open first relay:
`echo -e '\xA0\x01\01\xA2' > /dev/hidraw6`
Close first relay:
`echo -e '\xA0\x01\00\xA1' > /dev/hidraw6`

Perhaps this makes an specific command unnecessary.

darrylb123, 2023-04-18:
Cool, you've done the work for me. At least you have a work around until I
get mine and update usbrelay.
Just reading the link you supplied, it looks like it's another one that
can't be interrogated for current relay state.
Way back, I ran the original relays windows software in a VM with the USB
device mapped through then snooped the protocol with Wireshark.
The ucreatefun relays were dumb one direction protocol and the relay could
not be identified.
This one implies that the A0 identifier may be able to be changed.

alfem, 2023-04-18:
I can not see that one you can interrogate. I tested mine but I can not read anything from the device files. 

'A0' could some model prefix. I don't think it is useful for anything.

darrylb123, 2023-04-18:
Indeed, more rubbish.
I'm inclined to just document what you have done and leave it at that.
The documentation about using udev to apply unique device files for each
USB port for ucreatefun relays applies with these relays,

I asked the vendor about querying the state and changing the A0. We'll see
what they have to say.

alfem, 2023-04-19:
Meanwhile I have put together a little script for my Home Assistant server: 

https://github.com/alfem/usb-relay-control/

I asked the vendor yesterday as well, but he did not give me any info.

GoolaDev, 2023-06-26:
I've tried it and it works on Linux(Fedora). But I need it on Windows.
Using: hidapitester --vidpid 5131:2007 --open --send-output A0,01,01,A2
did nothing. There is a page option/switch (--usagePage ), but which one to use?

GoolaDev, 2023-06-27:
To ansert my own question. Here is LC HID Tester software for windows:
https://onedrive.live.com/?authkey=%21AO6nMeHgMBC3Bc8&id=4A0865B22350D05C%21125&cid=4A0865B22350D05C&parId=root&parQt=sharedby&o=OneUp

AMV007, 2024-01-03:
added functionality: https://github.com/darrylb123/usbrelay/pull/109
@alfem

closed 2024-01-07 by darrylb123

LupusTheCanine, 2024-01-13:
A bit more info on the device
Manufacturer site: http://www.chinalctech.com/index.html
Manufacturer wiki: http://wiki.lctech.cc/index.php?title=Main_Page

wlan4you, 2024-01-20:
> I've tried it and it works on Linux(Fedora). But I need it on Windows. Using: hidapitester --vidpid 5131:2007 --open --send-output A0,01,01,A2 did nothing. There is a page option/switch (--usagePage ), but which one to use?

I have absolutly the same problem and the same module.
i tryed it to switch with the same testtool and commands.
I will need it under windows for example with a batch file.

You find a solution over hidapitester ?