darrylb123/usbrelay #105: usbrelay_py installation & usage experience

hovercraft-github, 2023-09-16:
1. Looks like it is not compatible with python setuptools>=50, resulting in "ImportError: cannot import name 'setuptools' from 'setuptools'" error, so I had to edit usbrelay_py/pyproject.toml file and replace "setuptools>=42" with "setuptools==42" to be able to build the python library.
2. Appear that call to usbrelay_py.board_count() is mandatory to initialize the library, may be mention this in the doc?

darrylb123, 2023-09-17:
I have confirmed your bug report. 
I don't know enough about python to 'fix' the issue. A fix would be that it works with newer versions of setuptools. Your workaround will suffice until I can find an answer.
It appears that you need to call count() or details() first to initialise things. I'll add it to the doco.

darrylb123, 2023-09-17:
I've found the solution to building on later versions of setuptools. Trouble is, it is incompatible with versions < 49.
Setting is now >=49

closed 2023-09-17 by darrylb123