darrylb123/usbrelay #106: Raspberry pi 4 - python

ArashiCZ, 2023-09-26:
Hello.

I'm running 64bit raspios lite.
I tried to use pip3 install usbrelay-py, however i'm constantly getting this error:

` 
sudo pip3 install usbrelay-py
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting usbrelay-py
  Using cached usbrelay_py-1.0.tar.gz (4.1 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... error
  ERROR: Command errored out with exit status 1:
   command: /usr/bin/python3 /tmp/tmp60x1dyl7_in_process.py get_requires_for_build_wheel /tmp/tmpkxq_s836
       cwd: /tmp/pip-install-erk9tjr_/usbrelay-py_039f083e8680440686396df0d7a080d8
  Complete output (15 lines):
  Traceback (most recent call last):
    File "/tmp/tmp60x1dyl7_in_process.py", line 280, in <module>
      main()
    File "/tmp/tmp60x1dyl7_in_process.py", line 263, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/tmp/tmp60x1dyl7_in_process.py", line 114, in get_requires_for_build_wheel
      return hook(config_settings)
    File "/usr/local/lib/python3.9/dist-packages/setuptools/build_meta.py", line 355, in get_requires_for_build_wheel
      return self._get_build_requires(config_settings, requirements=['wheel'])
    File "/usr/local/lib/python3.9/dist-packages/setuptools/build_meta.py", line 325, in _get_build_requires
      self.run_setup()
    File "/usr/local/lib/python3.9/dist-packages/setuptools/build_meta.py", line 341, in run_setup
      exec(code, locals())
    File "<string>", line 17, in <module>
  ImportError: cannot import name 'setuptools' from 'setuptools' (/usr/local/lib/python3.9/dist-packages/setuptools/__init__.py)
  ----------------------------------------
WARNING: Discarding https://files.pythonhosted.org/packages/0c/ad/4036a83246a7f7e59919323fd4d1f589284b9afe539fe33c0a8036bd12f2/usbrelay_py-1.0.tar.gz#sha256=d0c74c358132ce1055487ebd6ad59bb84209d8417d670ff9d61c798eb8b783d1 (from https://pypi.org/simple/usbrelay-py/). Command errored out with exit status 1: /usr/bin/python3 /tmp/tmp60x1dyl7_in_process.py get_requires_for_build_wheel /tmp/tmpkxq_s836 Check the logs for full command output.
ERROR: Could not find a version that satisfies the requirement usbrelay-py
ERROR: No matching distribution found for usbrelay-py
`
Tried to reinstall setuptools, update pip, few other things and nothing.

wentasah, 2023-09-26:
Cannot you just `apt install python3-usbrelay`?

darrylb123, 2023-09-26:
> File "", line 17, in

Don't use pip. It's a binary library and is dependent on another binary
library. Use the GitHub source to build and install it.

ArashiCZ, 2023-09-27:
> Cannot you just `apt install python3-usbrelay`?

Apparently not.
[image]

ArashiCZ, 2023-09-27:
> Don't use pip. It's a binary library and is dependent on another binary library. Use the GitHub source to build and install it.
> […](#)

This is what i get when running the "make" command inside the folder usbrelay_py.
```
pi@juhu:~/python_tests/usbrelay/usbrelay-master/usbrelay_py $ make
make: *** No rule to make target '../libusbrelay.so.1.1', needed by 'all'.  Stop.
```

darrylb123, 2023-09-27:
I'd be careful with the Debian package. It's not maintained.

mefuller, 2023-09-27:
The Fedora package is fully up-to-date!

darrylb123, 2023-09-27:
That's odd. There is a Debian python3-usbrelay package that is the current
version. Can you just install usbrelay package?

darrylb123, 2023-09-27:
Ha, almost. Did you see my mastodon post to you and Michal :)

ArashiCZ, 2023-09-27:
I tried to use atleast the "Sudo usbrelay" that i installed with apt, which works perfectly by the way with "| sed" and locating lines (relay) after the line with "Number of Relays" but it behaves strangly. It doesn't catch the line with the serial numbers.

I would like to avoid this by using the python extension.
Would someone help me a guide me how to build the library for python.
I'm planing to use it my upcoming project that is build upon these usb relays (home automation),
is it possible to just copy the library after building it to differnt pi ? (same hw).

Thank you.

mefuller, 2023-09-27:
@darrylb123 I did see your message, which is why v1.2.1 is in Rawhide, and within a couple days of clearing testing on F39 - F37: https://src.fedoraproject.org/rpms/usbrelay

darrylb123, 2023-09-27:
The documentation to build usbrelay and python extension is in the readme
on GitHub.
If you installed the package, you should be able to check the manual page.
Also, the option flags are output with the -h flag:
$ usbrelay -h
usbrelay: invalid option -- 'h'
Try `usbrelay --help' or `usbrelay --usage' for more information.
***@***.*** ~]$ usbrelay --help
Usage: usbrelay [OPTION...] [ACTION...]
Control or query USB HID relays.

  -d, --debug                Produce debugging output
  -e, --export-id=DEV        Print relay ID of the given DEV
(/dev/hidrawXX) in
                             udev-compatible format
  -q, --quiet                Be quiet
  -?, --help                 Give this help list
      --usage                Give a short usage message
  -V, --version              Print program version

Mandatory or optional arguments to long options are also mandatory or
optional
for any corresponding short options.

Without ACTION, the actual state of all relays is printed to stdout.
ACTION can be one of:
RELID_N=[0|1] to switch the N-th relay off or on
RELID_0=NEWID to change relay ID

Report bugs to https://github.com/darrylb123/usbrelay/issues.

ArashiCZ, 2023-09-27:
> The documentation to build usbrelay and python extension is in the readme on GitHub. If you installed the package, you should be able to check the manual page. Also, the option flags are output with [...]

Yes, I looked at the help, but as i posted before, I cannot build the library according to the instructions.
`pi@juhu:~/python_tests/usbrelay/usbrelay-master/usbrelay_py $ make
make: *** No rule to make target '../libusbrelay.so.1.1', needed by 'all'.  Stop.`

darrylb123, 2023-09-27:
You have to build the main library first.
Run make in the top level directory first then sudo make install, then cd
to usbrelay_py and run make and sudo make install.

ArashiCZ, 2023-09-27:
> You have to build the main library first. Run make in the top level directory first then sudo make install, then cd to usbrelay_py and run make and sudo make install.
> […](#)

Oh, I'm sorry @darrylb123 . That totaly went over my head,
Will do that asap and let you know.

Thank you.

closed 2023-09-27 by ArashiCZ