darrylb123/usbrelay #107: Relay serial number contains undefined character

shimi25, 2023-11-07:
Hi , 
I've compiled the code successfully. 
However , when I'm addressing the relay with the `sudo usbrelay -d ` command I'm getting a character in the serial number and in the port numbers that I don't know how to use. The serial I'm getting is :    serial_number: ▒QMBS 
Here is a screen shot :

[image]

Someone has any idea how to solve this ?

Thanks , 

Shimon.

darrylb123, 2023-11-07:
Change the serial.
You can use the device usbrelay /dev/hidraw0_0=ABCDE

Check out the doco

Darryl

shimi25, 2023-11-09:
Thank you Darryl !

closed 2023-11-09 by shimi25

GarwinCecil, 2024-03-18:
> Change the serial. You can use the device usbrelay /dev/hidraw0_0=ABCDE Check out the doco Darryl
> […](#)

hi Darry, the log says that:
   Invalid relay specification: /dev/hidraw3=D0010

GarwinCecil, 2024-03-18:
CMD: usbrelay /dev/hidraw3_0=D0010
now its ok