darrylb123/usbrelay #108: USB Relay MQTT service won't start after upgrade to Debian 11 bullseye

dezanche, 2023-11-08:
Service ran fine for almost 2 years under Debian 10 (although I had to start it manually after each reboot) but I cannot get it to start after upgrading to Debian 11.
Tried both my old `usbrelayd.service` file as well as the updated one in the repo.
Here is what `systemctl status usbrelayd` says:
> * usbrelayd.service - USB Relay MQTT service
     Loaded: loaded (/etc/systemd/system/usbrelayd.service; enabled; vendor preset: enabled)
     Active: failed (Result: exit-code) since Tue 2023-11-07 21:12:20 MST; 9s ago
    Process: 23154 ExecStart=/usr/bin/python3 /usr/local/sbin/usbrelayd ${BROKER} (code=exited, status=1/FAILURE)
   Main PID: 23154 (code=exited, status=1/FAILURE)

I tried upgrading the whole package to a newer version but `apt-get install usbrelay` tells me `usbrelay is already the newest version (0.4-2).`

darrylb123, 2023-11-08:
That version is ancient. Debian packages are a bit hit and miss.
The best way is to build from the github source. The documentation
describes how to do that.
You should replace the usbrelayd.service with the latest too.
Remove the usbrelay package first (apt-get remove usbrelay).

wentasah, 2023-11-08:
Why using Debian 11? Debian 12 has a newer version that should work well. If you really need Debian 11, than Darryl's suggestion is a way to go.

dezanche, 2023-11-08:
Thank you both.
I'm running Openmediavault which has not been released on Debian 12 yet. My home automation software is all running in docker containers so I may do that for this as well. Which do you recommend?

darrylb123, 2023-11-08:
Build from GitHub source. There is a docker build file there, but I don't
know how good it is.

dezanche, 2023-11-09:
I built as described and usbrelayd worked right away after I set the correct URL for the MQTT server.
One question did come up while reading the installation instructions: under what circumstances would I need to add users? Luckily I didn't have to (possibly because I had it already set up from the old installation) but I would've been scratching my head if I did.
Thanks for the help!

closed 2023-11-14 by dezanche