darrylb123/usbrelay #110: MQTT commands stopped working

dezanche, 2024-01-25:
I'm not sure if this is related to my previous issue (#108), but after a fresh rebuild it worked correctly for only a few days, then MQTT functionality broke after a reboot.
I can still query the state and set relays from the shell using `usbrelay` commands, so the HW and USB connection are fine.
However, MQTT commands from Home Assistant are visible using MQTT Explorer but they don't trigger the relays. The service is running:
```
root@rock64:/home/home-assistant# systemctl status usbrelayd
* usbrelayd.service - USB Relay MQTT service
     Loaded: loaded (/etc/systemd/system/usbrelayd.service; enabled; vendor preset: enabled)
     Active: active (running) since Wed 2024-01-24 21:43:59 MST; 7min ago
       Docs: man:usbrelayd(8)
   Main PID: 11129 (python3)
      Tasks: 2 (limit: 4772)
     Memory: 11.3M
     CGroup: /system.slice/usbrelayd.service
             -11129 /usr/bin/python3 /usr/sbin/usbrelayd
Jan 24 21:51:01 rock64 python3[11129]: State:  stat//4 OFF
Jan 24 21:51:01 rock64 python3[11129]: Subscribed:  cmnd//4
Jan 24 21:51:01 rock64 python3[11129]: State:  stat//5 OFF
Jan 24 21:51:01 rock64 python3[11129]: Subscribed:  cmnd//5
Jan 24 21:51:01 rock64 python3[11129]: State:  stat//6 OFF
Jan 24 21:51:01 rock64 python3[11129]: Subscribed:  cmnd//6
Jan 24 21:51:01 rock64 python3[11129]: State:  stat//7 OFF
Jan 24 21:51:01 rock64 python3[11129]: Subscribed:  cmnd//7
Jan 24 21:51:01 rock64 python3[11129]: State:  stat//8 OFF
Jan 24 21:51:01 rock64 python3[11129]: Subscribed:  cmnd//8
```
It's suspicious, though, that relays 1-3, which are the ones I actually use (and I can still set using `usbrelay` commands), are not showing up on this list.
I can't figure this out so any help is appreciated.

darrylb123, 2024-01-25:
Notice there is no topic name in the log. It is subscribing to an empty
name.
The topic name is read from the relay. What is the output of sudo usbrelay

darrylb123, 2024-01-25:
It should look something like this:
Jan 25 16:08:08 raspberrypi python3[421]: Subscribed:  cmnd/WATER/5
Jan 25 16:08:08 raspberrypi python3[421]: State:  stat/WATER/6 OFF
Jan 25 16:08:08 raspberrypi python3[421]: Subscribed:  cmnd/WATER/6
Jan 25 16:08:08 raspberrypi python3[421]: State:  stat/WATER/7 OFF
Jan 25 16:08:08 raspberrypi python3[421]: Subscribed:  cmnd/WATER/7
Jan 25 16:08:08 raspberrypi python3[421]: State:  stat/WATER/8 OFF
Jan 25 16:08:08 raspberrypi python3[421]: Subscribed:  cmnd/WATER/8
Jan 25 16:08:08 raspberrypi python3[421]: received message:  cmnd/WATER/6
OFF
Jan 25 16:08:09 raspberrypi python3[421]: received message:  cmnd/WATER/7
OFF
Jan 25 16:08:09 raspberrypi python3[421]: received message:  cmnd/WATER/8
OFF

 $ sudo usbrelay
WATER_1=0
WATER_2=0
WATER_3=0
WATER_4=0
WATER_5=0
WATER_6=0
WATER_7=0
WATER_8=0

dezanche, 2024-01-25:
I pasted below, and I don't see anything wrong there. I'll add also that I tried restarting the service as well as rebooting and nothing changed.
```
root@rock64:~# usbrelay -d
libusbrelay: 1.2.1
usbrelay: 1.2.1
enumerate_relay_boards()Found 1 devices
Device Found
  type: 16c0 05df
  path: /dev/hidraw0
  serial_number: 6QMBS
Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
  Module_type = 1
6QMBS_1=0
6QMBS_2=0
6QMBS_3=0
6QMBS_4=0
6QMBS_5=0
6QMBS_6=0
6QMBS_7=0
6QMBS_8=0
```
Thanks,
N

darrylb123, 2024-01-25:
Is see you're running as root, does the user that usbrelay runs as have
permission to the device. Is it in the (plugdev on debian) group?

dezanche, 2024-01-26:
Here is what groups shows:
```
root@rock64:~# groups usbrelayd
usbrelayd : usbrelayd
root@rock64:~# groups usbrelay
usbrelay : usbrelay plugdev
```
I did miss the change required for Debian in `50-usbrelay.rules` but after updating and restarting the daemon nothing changes.

darrylb123, 2024-01-26:
You need to reboot or get udev to reload the rules. Restarting usbrelayd
won't help.

dezanche, 2024-01-27:
Actually I read that to reload the udev rules it's enough to unplug the USB connection and plug it back in, and it worked!
Thanks and sorry for all the noob questions. Last time I worked this deep in a linux system USB had barely been invented...
I almost burnt my finger on the voltage regulator of that board so I have to deal with that now

closed 2024-01-27 by dezanche