darrylb123/usbrelay #113: issue with renaming

toby11, 2024-05-18:
since updating to Ubuntu 22.04.4 LTS, usbrelay is returning back bad characters in the device names

usbrelay is already the newest version (0.4-2).

toby@homeassistant:~$ usbrelay
Device Found
  type: 16c0 05df
  path: /dev/hidraw1
  serial_number:
  Manufacturer: www.dcttech.com
  Product:      USBRelay8
  Release:      100
  Interface:    0
  Number of Relays = 8
▒▒▒▒▒_1=0
▒▒▒▒▒_2=0
▒▒▒▒▒_3=0
▒▒▒▒▒_4=0
▒▒▒▒▒_5=0
▒▒▒▒▒_6=0
▒▒▒▒▒_7=0
▒▒▒▒▒_8=0

if I try and rename is as per the docs

darrylb123, 2024-05-18:
That version is ancient.
If your package is the newest available to install, you should remove it
and build the latest version from the source on GitHub.

toby11, 2024-05-18:
yes thats fixed it- sorry should have tried that first.

closed 2024-05-18 by toby11