darrylb123/usbrelay #114: Feature Request: re-program product string.

mjp66, 2024-06-14:
This is not a bug report, but a feature request.

I purchased two 2-channel relay boards which arrived with the same serial number.
I used your program on a Raspberry Pi to rename one of them.
I later purchased two more units, and these units had unintelligible serial numbers, which your program was also able to fix.
But the second set of relays have an additional problem: they claim to be 4 channel units instead of 2 channels, i.e. report USBRelay4.

Do you think you might be able to reprogram this 'product' string?

Thanks for a great program,
-Mike

darrylb123, 2024-06-17:
That was nice of them.
Does ignoring 3 & 4 cause any issues?
The firmware is open source, I wonder if it's possible to replace the
firmware with the correct count.

darrylb123, 2024-06-17:
https://github.com/thmahe/usbrelay-firmware

mjp66, 2024-06-17:
Thanks for the link, I’ll look that over.

closed 2024-06-17 by mjp66

darrylb123, 2024-06-21:
Can you leave a comment on success or failure. I'd be very interested to know.