darrylb123/usbrelay #115: Querying the state of Line

ostasevych, 2024-06-30:
Hi! I am looking for the way to detect power outage in the circuit. Can I use this software code with the usb relay like https://www.aliexpress.com/item/1005006781536777.html for this goal? I mean, I will connect plug to the relay, plug it to the mains line. When there's no power it will send me the alert about that?

darrylb123, 2024-06-30:
I don't understand what you are trying to do. Can you attach a sketch of
the circuit you propose?

ostasevych, 2024-07-01:
Quite simple. I would like to detect if there's power in the 220 V circuit, and to send alerts when it appears and disappears.

darrylb123, 2024-07-01:
Ok, then this won't help you. It does not detect voltage at all. A 220v
relay with the supply connected to the relay coil, with an esp8266 running
tasmota monitoring a relay contact might be closer to what you're looking
for.

ostasevych, 2024-07-01:
> Ok, then this won't help you. It does not detect voltage at all. A 220v
> relay with the supply connected to the relay coil, with an esp8266 running
> tasmota monitoring a relay contact might be closer to what you're looking
> for.
> 
> 
> > Quite simple. I would like to detect if there's power in the 220 V
> > circuit, and to send alerts when it appears and disappears.

Oh, thanks for the quick reply. May I ask you to send examples of devices/pcb which have this feature?