darrylb123/usbrelay #121: Change relay "_1=1" , but still show "_1=0"

Lucy0418s, 2025-02-18:
I have a question... 
I install usbrelay with latest source code build.

I set usbrelay _1 = 1, the command is take effect on the relay, but "usbrelay" still show _1=0...

[image]

[image]

same issue with change name of relay:

[image]

@darrylb123 can you help me for this issue?

darrylb123, 2025-02-18:
You'll need to use the device (/dev/hidrawx) to set the name since it doesn't have one. How to do that is in the documentation.

Lucy0418s, 2025-02-18:
[image]
@darrylb123 I tried to use device to set name, but still not work..

darrylb123, 2025-02-18:
Module type 3 is an LCUS relay that are very brain dead.
I haven't played with one in a long time but I think you need to control it via the device. You cannot set the serial.

darrylb123, 2025-02-18:
It's also not possible to query the relay state on an LCUS relay. All you can do is operate it via the device.

Lucy0418s, 2025-02-19:
@darrylb123  if that, I wonder which module type relay support all functions including query relay state and set serial?

darrylb123, 2025-02-19:
Only the DCTTECH relays support all functionality.
They are quite common and are usually are labelled USB-Relay-x.
They also supply application C++ code and windows DLL.
LCUS and UCREATEFUN relays simply take an ASCII string and can only be turned on or off.

closed 2025-05-08 by darrylb123