darrylb123/usbrelay #124: invalid #020 char on topics for MQTT

javajetcoder, 2025-05-03:
Thanks for this great software. Install py module on Pi Zero W inserts invalid #020 char on topics for MQTT. Any idea why?

2025-05-03T09:29:30.018078-04:00 garden python3[913]: MQTT Broker:  **** (redacted)
2025-05-03T09:29:30.019033-04:00 garden python3[913]: MQTT Client:    **** (redacted)
2025-05-03T09:29:30.019444-04:00 garden python3[913]: Modules Connected:  1
2025-05-03T09:29:30.062793-04:00 garden python3[913]: State:  stat/6QM#020S/1 OFF
2025-05-03T09:29:30.063315-04:00 garden python3[913]: Subscribed:  cmnd/6QM#020S/1
2025-05-03T09:29:30.063628-04:00 garden python3[913]: State:  stat/6QM#020S/2 OFF
2025-05-03T09:29:30.063875-04:00 garden python3[913]: Subscribed:  cmnd/6QM#020S/2
2025-05-03T09:29:30.064134-04:00 garden python3[913]: State:  stat/6QM#020S/3 OFF
2025-05-03T09:29:30.064377-04:00 garden python3[913]: Subscribed:  cmnd/6QM#020S/3
2025-05-03T09:29:30.064627-04:00 garden python3[913]: State:  stat/6QM#020S/4 OFF
2025-05-03T09:29:30.064855-04:00 garden python3[913]: Subscribed:  cmnd/6QM#020S/4
2025-05-03T09:29:30.065058-04:00 garden python3[913]: State:  stat/6QM#020S/5 OFF
2025-05-03T09:29:30.065252-04:00 garden python3[913]: Subscribed:  cmnd/6QM#020S/5
2025-05-03T09:29:30.065448-04:00 garden python3[913]: State:  stat/6QM#020S/6 OFF
2025-05-03T09:29:30.065740-04:00 garden python3[913]: Subscribed:  cmnd/6QM#020S/6
2025-05-03T09:29:30.066030-04:00 garden python3[913]: State:  stat/6QM#020S/7 OFF
2025-05-03T09:29:30.066301-04:00 garden python3[913]: Subscribed:  cmnd/6QM#020S/7
2025-05-03T09:29:30.066492-04:00 garden python3[913]: State:  stat/6QM#020S/8 OFF
2025-05-03T09:29:30.066728-04:00 garden python3[913]: Subscribed:  cmnd/6QM#020S/8

darrylb123, 2025-05-08:
I think the current serial set in the module has a non-ascii character.
Try using usbrelay to change the serial to what you expect.

javajetcoder, 2025-05-08:
yep.

closed 2025-05-26 by darrylb123